stupidloud/nanopi-openwrt #572: R1plus使用闪迪ultraTF卡无法正常启动

LCW0NJUPT, 2021-04-22:
### 一、问题描述：无法启动该镜像
硬件：R1Plus
固件版本： 2021-04-21 r1p-with-docker.img.gz
TF卡：闪迪ultra16GB
测试了官方的openwrt镜像是可以正常启动运行的，排除了硬件故障的可能，因此来这里反馈

> 很想知道是不是固件兼容性的问题

### 二、 以下为串口打印的开机日志

```cs
DDR version 1.16 20190528
ID:0x805 N
In
DDR4
333MHz
Bus Width=32 Col=10 Bank=4 Bank Group=2 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
ddrconfig:14
OUT
Boot1 Release Time: May 13 2019 17:34:36, version: 2.50
ChipType = 0x11, 248
mmc2:cmd1,20
emmc reinit
mmc2:cmd1,20
emmc reinit
mmc2:cmd1,20
SdmmcInit=2 1
mmc0:cmd5,20
SdmmcInit=0 0
BootCapSize=0
UserCapSize=15193MB
FwPartOffset=2000 , 0
StorageInit ok = 33223
Raw SecureMode = 0
SecureInit read PBA: 0x4
SecureInit read PBA: 0x404
SecureInit read PBA: 0x804
SecureInit read PBA: 0xc04
SecureInit read PBA: 0x1004
SecureInit ret = 0, SecureMode = 0
atags_set_bootdev: ret:(0)
GPT 0x337a9f0 signature is wrong
recovery gpt...
GPT 0x337a9f0 signature is wrong
recovery gpt fail!
LoadTrust Addr:0x4000
No find bl30.bin
No find bl32.bin
Load uboot, ReadLba = 2000
hdr 000000000337a3b0 + 0x0:0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,

Load OK, addr=0x200000, size=0xa5958
RunBL31 0x40000
INFO:    Preloader serial: 2
NOTICE:  BL31: v1.3(debug):403e0b816
NOTICE:  BL31: Built : 14:13:08, Aug 11 2020
NOTICE:  BL31:Rockchip release version: v1.3
INFO:    ARM GICv2 driver initialized
INFO:    Using opteed sec cpu_context!
INFO:    boot cpu mask: 1
INFO:    plat_rockchip_pmu_init: pd status 0xe
INFO:    BL31: Initializing runtime services
WARNING: No OPTEE provided by BL2 boot loader, Booting device without OPTEE initialization. SMC`s destined for OPTEE will return SMC_UNK

ERROR:   Error initializing runtime service opteed_fast

INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9

[... 247 lines cut ...]
[    0.654735] usb 4-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    0.992927] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    1.493461] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    1.496289] rk808-rtc rk808-rtc: setting system clock to 2016-01-21T08:52:11 UTC (1453366331)
[    1.498392] Waiting for root device PARTUUID=5452574f-02...

[    1.530456] mmc0: error -110 whilst initialising SD card

[    2.031822] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    2.532353] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    2.546995] mmc_host mmc0: Bus speed (slot 0) = 300000Hz (slot req 300000Hz, actual 300000HZ div = 0)
[    3.076233] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    3.576766] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    3.591248] mmc_host mmc0: Bus speed (slot 0) = 200000Hz (slot req 200000Hz, actual 200000HZ div = 0)
[    4.128713] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    4.629248] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[    4.642729] mmc_host mmc0: Bus speed (slot 0) = 100000Hz (slot req 100000Hz, actual 100000HZ div = 0)
[    5.200313] dwmmc_rockchip ff500000.dwmmc: Busy; trying anyway
[    5.700844] mmc_host mmc0: Timeout sending command (cmd 0x202000 arg 0x0 status 0x80202000)
[   31.714760] vcc_sd: disabling

```
### 三、猜测
#### 看上述报错信息，似乎是TF卡没有被正确加载，进而怀疑是RK3328不兼容该TF卡。但随后测试的orangepi官方openwrt以及debain均能正常boot进入系统和正常工作，因此此项怀疑不成立，那么是镜像自身的兼容性问题吗？
**欢迎大家讨论！**

Engrtao, 2021-04-25:
问题解决了，我也是这样的情况

LCW0NJUPT, 2021-04-25:
> 问题解决了，我也是这样的情况

请问您是如何解决的呢？我昨晚尝试了24号的版本，依然是这个报错的。

lituo666, 2021-05-04:
我也是遇到这个问题了，请问大佬们怎么解决呢？

LCW0NJUPT, 2021-05-04:
> 我也是遇到这个问题了，请问大佬们怎么解决呢？

我更换了凯侠的TF卡就行了，这个固件似乎和闪迪的卡不兼容。

lituo666, 2021-05-04:
> > 我也是遇到这个问题了，请问大佬们怎么解决呢？
> 
> 我更换了凯侠的TF卡就行了，这个固件似乎和闪迪的卡不兼容。
谢谢，我试试

lituo666, 2021-05-04:
我试了 夏科 的TF卡，可以使用

lituo666, 2021-05-11:
> > 我试了 夏科 的TF卡，可以使用
> 
> 请问你用的是哪款设备呢，我用的是orangepi R1plus，不兼容的现象很有趣，想看一下是不是只有这一款设备出现这种情况

也是orangepi R1 plus ，我之前用到[Lean](https://github.com/coolsnowwolf/lede)的固件使用闪迪tf卡，夏科的卡就不行。现在换这个固件就只能用夏科的tf卡，用闪迪的tf卡和你的情况是一样的。

isyekong, 2022-07-29:
同样的问题，闪迪卡无法启动