stupidloud/nanopi-openwrt #575: R2S做主路由后，speedtest.net测速大幅下降

caesaranthony, 2021-04-23:
家里目前是移动千兆宽带，光猫桥接路由器拨号上网；之前用硬路由，有线测速大概五六百左右
换R2S做路由器，刷了大神的固件
ImmortalWrt 18.06-SNAPSHOT r9439-666852252 /LuCI openwrt-18.06 branch (git-21.085.39515-b0e986b)
R2S直连电脑测速，没有在通过交换机，网络结构跟原来硬路由一样
测速大部分时间只有一百左右了，偶尔能到二三百
请大神说说对此现象的看法，谢谢。
另外，还有一个现象；使用aria2插件下载，速度二三兆，R2S的cpu使用率能达到30-40%，温度60℃以上，您看这正常么？
本人坐标北京，目前气温20℃左右。

caesaranthony, 2021-04-23:
看到之前有讨论过掉速问题，有可能是开启了open****插件的原因
我并没有使用这个插件，但是也掉速。

wei727, 2021-04-23:
我也遇到过掉速的情况，用一段时间本地带宽就会掉速。重启后正常用一段时间又开始掉速，我暂时更换其它固件了。

caesaranthony, 2021-04-23:
> 我也遇到过掉速的情况，用一段时间本地带宽就会掉速。重启后正常用一段时间又开始掉速，我暂时更换其它固件了。

你现在用谁编译的固件了？稳定么？

wei727, 2021-04-23:
> > 我也遇到过掉速的情况，用替换本地更多就会掉速。重启后正常用重启又开始掉速，我暂时更换其他固件了。
> 
> 你现在用谁编译的固件了？稳定么？

404大佬和骷髅头的固件

caesaranthony, 2021-04-23:
> > > 我也遇到过掉速的情况，用替换本地更多就会掉速。重启后正常用重启又开始掉速，我暂时更换其他固件了。
> > 
> > 
> > 你现在用谁编译的固件了？稳定么？
> 
> 404大佬和骷髅头的固件

没再掉速？

wei727, 2021-04-23:
> > > 我也遇到过掉速的情况，用替换本地更多就会掉速。重启后正常用重启又开始掉速，我暂时更换其他固件了。
> > 
> > 
> > 你现在用谁编译的固件了？稳定么？
> 
> 404大佬和骷髅头的固件

> > > > 我也遇到过掉速的情况，用替换本地更多就会掉速。重启后正常用重启又开始掉速，我暂时更换其他固件了。
> > > 
> > > 
> > > 你现在用谁编译的固件了？稳定么？
> > 
> > 
> > 404大佬和骷髅头的固件
> 
> 没再掉速？

404和骷髅头我能保证本地带宽不掉速，我前前后后也装了差不多10个左右固件了。至于其它还要看你用的插件，有些固件核心更新得勤插件很多不怎么更新，很多掉速是插件造成的。

caesaranthony, 2021-04-23:
> > > > 我也遇到过掉速的情况，用替换本地更多就会掉速。重启后正常用重启又开始掉速，我暂时更换其他固件了。
> > > 
> > > 
> > > 你现在用谁编译的固件了？稳定么？
> > 
> > 
> > 404大佬和骷髅头的固件
> 
> > > > > 我也遇到过掉速的情况，用替换本地更多就会掉速。重启后正常用重启又开始掉速，我暂时更换其他固件了。
> > > > 
> > > > 
> > > > 你现在用谁编译的固件了？稳定么？
> > > 
> > > 
> > > 404大佬和骷髅头的固件
> > 
> > 
> > 没再掉速？
> 
> 404和骷髅头我能保证本地带宽不掉速，我前前后后也装了差不多10个左右固件了。至于其它还要看你用的插件，有些固件核心更新得勤插件很多不怎么更新，很多掉速是插件造成的。

多谢及时回复

Hiccup6, 2021-04-23:
别用openclash 流量都走他，CPU负载太高了

caesaranthony, 2021-04-23:
> 别用openclash 流量都走他，CPU负载太高了

没用这个插件

caesaranthony, 2021-04-24:
我修改了一下系统配置
我的TF卡是64G的，装入系统后，还有很大的剩余空间，我就把剩余空间单独分区作为一个挂载点挂载系统中，用于存储一些临时文件。昨天我把此挂载点卸载，删除分区，期间重启了系统，之后测试网速，是可以跑满带宽的。
之后我用aria2挂了一夜BT，并且还进行了其他操作；今天间隔三四个小时仍然可以跑满带宽。
我在想，之前网络降速，是否与我将TF卡剩余空间单独划分出来的操作有关；我会持续不定时进行测试，看看系统是否还会降速。

caesaranthony, 2021-04-25:
降速现象好像是受“网络共享”功能影响
我从路由器外挂硬盘里通过共享功能，把下载好的电影拷贝到电脑上之后再次测速
网速就变成百兆左右了

wei727, 2021-04-25:
> 降速现象好像是受“网络共享”功能影响
> 我从路由器外挂硬盘里通过共享功能，把下载好的电影拷贝到电脑上之后再次测速
> 网速就变成百兆左右了

找到原因了吗？确定是这个插件吗！卸载或者禁用试试，如果是这个原因我空了刷回来也试试。

caesaranthony, 2021-04-25:
> > 降速现象好像是受“网络共享”功能影响
> > 我从路由器外挂硬盘里通过共享功能，把下载好的电影拷贝到电脑上之后再次测速
> > 网速就变成百兆左右了
> 
> 找到原因了吗？确定是这个插件吗！卸载或者禁用试试，如果是这个原因我空了刷回来也试试。

是“网络共享”功能的可能性很大，刚才突然瞬间断网，然后又恢复正常，可能是系统某个程序重置了
再次测试网速，恢复满带宽正常了
期间我没有手工重启系统

zane-ng, 2021-04-25:
我之前也是用 R2s 做主路由，宽带为中国电信下行 500Mbps，上行 50Mbps. Speedtest 直连测速可以跑满，甚至在 600Mbps 左右。
出现不能跑满的情况，我个人经验有两种：
1）OpenClash 仅使用 TUN 模式。应使用 TUN 混合模式（TCP - 转发，UDP - TUN）；
2）QoS 设置出现错误，人为进行了限速。

wei727, 2021-04-25:
> 我之前也是用 R2s 做主路由，宽带为中国电信下行 500Mbps，上行 50Mbps. Speedtest 直连测速可以跑满，甚至在 600Mbps 左右。
> 出现不能跑满的情况，我个人经验有两种：
> 1）OpenClash 仅使用 TUN 模式。应使用 TUN 混合模式（TCP - 转发，UDP - TUN）；
> 2）QoS 设置出现错误，人为进行了限速。

我一般不用OpenClash太复杂，大家都是资源占用大CPU负载大。QoS这个东西我开都没开，我一般就开hello world + AdGuardHome +smartdns。同样这套组合在其它固件没有出现类似问题！

wei727, 2021-04-25:
> > > 降速现象好像是受“网络共享”功能影响
> > > 我从路由器外挂硬盘里通过共享功能，把下载好的电影拷贝到电脑上之后再次测速
> > > 网速就变成百兆左右了
> > 
> > 
> > 找到原因了吗？确定是这个插件吗！卸载或者禁用试试，如果是这个原因我空了刷回来也试试。
> 
> 是“网络共享”功能的可能性很大，刚才突然瞬间断网，然后又恢复正常，可能是系统某个程序重置了
> 再次测试网速，恢复满带宽正常了
> 期间我没有手工重启系统

其实这几个大佬都是用的同样源代码编译的固件，固件本身应该没有大问题很大可能就是插件的原因只是不知道是那个插件造成的。我一般装好固件后我不用的插件都会卸载不用的功能都会关闭，就是为了避免出现这种情况。

caesaranthony, 2021-04-26:
今天用aria2挂了一会儿bt，测试网速又下降了
重启路由器后恢复正常

wei727, 2021-04-26:
> 今天用aria2挂了一会儿bt，测试网速又下降了
> 重启路由器后恢复正常

我早上又装回1988的固件了，暂时现在一切正常。我不像你要挂下载这些，我把许多我不用的插件卸载和功能都关闭了。

neozhu, 2022-07-01:
我也试了，就开了wallpass一个，上国内网最快200Mbps/国外100Mbps（可能机场不给力）, 不走R2s路由 可以350Mbps。感觉性能是不行

jowu598, 2022-09-14:
m