stupidloud/nanopi-openwrt #582: ip数量巨大

lhf19900402, 2021-04-24:
ip数量巨大  我实际也就20个设备左右  ip列表里好多169开头的ip   为啥？？
[image]

169.254.42.248 | 08:ea:40:5a:a6:2b | br-lan
-- | -- | --
169.254.88.175 | 08:ea:40:5a:a6:2b | br-lan
169.254.40.23 | 08:ea:40:5a:a6:2b | br-lan
169.254.72.174 | 08:ea:40:5a:a6:2b | br-lan
169.254.65.218 | 08:ea:40:5a:a6:2b | br-lan
169.254.10.121 | 08:ea:40:5a:a6:2b | br-lan
169.254.49.100 | 08:ea:40:5a:a6:2b | br-lan
169.254.30.243 | 08:ea:40:5a:a6:2b | br-lan
169.254.71.77 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.230 | 54:75:95:d2:f7:95 | br-lan
169.254.70.102 | 08:ea:40:5a:a6:2b | br-lan
169.254.14.15 | 08:ea:40:5a:a6:2b | br-lan
169.254.58.222 | 08:ea:40:5a:a6:2b | br-lan
169.254.120.23 | 08:ea:40:5a:a6:2b | br-lan
169.254.109.252 | 08:ea:40:5a:a6:2b | br-lan
169.254.123.151 | 08:ea:40:5a:a6:2b | br-lan
169.254.9.242 | 08:ea:40:5a:a6:2b | br-lan
169.254.121.204 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.211 | 08:ea:40:53:45:09 | br-lan
192.168.2.231 | 08:ea:40:53:db:1c | br-lan
169.254.82.139 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.241 | 5c:e5:0c:0d:ed:3e | br-lan
169.254.19.9 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.157 | a4:14:37:bf:a6:8f | br-lan
169.254.63.210 | 08:ea:40:5a:a6:2b | br-lan
169.254.82.202 | 08:ea:40:5a:a6:2b | br-lan
169.254.40.31 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.200 | 78:8b:2a:46:0e:c2 | br-lan
169.254.122.189 | 08:ea:40:5a:a6:2b | br-lan
169.254.52.73 | 08:ea:40:5a:a6:2b | br-lan
169.254.58.167 | 08:ea:40:5a:a6:2b | br-lan
169.254.101.119 | 08:ea:40:5a:a6:2b | br-lan
169.254.107.194 | 08:ea:40:5a:a6:2b | br-lan
169.254.49.249 | 08:ea:40:5a:a6:2b | br-lan
169.254.43.46 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.119 | a0:c5:f2:b2:c8:8a | br-lan
192.168.2.248 | 6c:21:a2:bd:a2:cc | br-lan
169.254.45.42 | 08:ea:40:5a:a6:2b | br-lan
169.254.3.127 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.144 | 24:5b:a7:ba:55:49 | br-lan
192.168.2.162 | 44:23:7c:2e:c5:20 | br-lan
169.254.73.138 | 08:ea:40:5a:a6:2b | br-lan
169.254.42.251 | 08:ea:40:5a:a6:2b | br-lan
169.254.77.250 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.197 | a2:28:27:e9:be:f9 | br-lan
169.254.34.194 | 08:ea:40:5a:a6:2b | br-lan
169.254.111.128 | 08:ea:40:5a:a6:2b | br-lan
169.254.88.176 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.221 | f4:5c:89:92:19:15 | br-lan
169.254.125.179 | 08:ea:40:5a:a6:2b | br-lan
169.254.37.203 | 08:ea:40:5a:a6:2b | br-lan
169.254.2.210 | 08:ea:40:5a:a6:2b | br-lan
169.254.74.196 | 08:ea:40:5a:a6:2b | br-lan
169.254.55.222 | 08:ea:40:5a:a6:2b | br-lan
169.254.6.214 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.135 | 50:ec:50:8c:cb:94 | br-lan
169.254.6.95 | 08:ea:40:5a:a6:2b | br-lan
169.254.36.7 | 08:ea:40:5a:a6:2b | br-lan
169.254.29.77 | 08:ea:40:5a:a6:2b | br-lan
169.254.19.17 | 08:ea:40:5a:a6:2b | br-lan
169.254.86.155 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.163 | ae:5d:8c:77:7a:f9 | br-lan
169.254.89.148 | 08:ea:40:5a:a6:2b | br-lan
169.254.40.1 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.183 | 08:ea:40:5a:a6:2b | br-lan
169.254.65.87 | 08:ea:40:5a:a6:2b | br-lan
169.254.19.180 | 08:ea:40:5a:a6:2b | br-lan
169.254.58.137 | 08:ea:40:5a:a6:2b | br-lan
169.254.46.105 | 08:ea:40:5a:a6:2b | br-lan
169.254.179.136 | 00:db:df:30:8d:fb | br-lan
169.254.119.201 | 08:ea:40:5a:a6:2b | br-lan
169.254.89.53 | 08:ea:40:5a:a6:2b | br-lan
192.168.2.202 | 50:d2:f5:5b:0c:9c | br-lan
169.254.50.89 | 08:ea:40:5a:a6:2b | br-lan
169.254.120.95 | 08:ea:40:5a:a6:2b | br-lan
169.254.127.174 | 08:ea:40:5a:a6:2b | br-lan
169.254.30.0 | 08:ea:40:5a:a6:2b | br-lan

klever1988, 2021-04-24:
查一下08:ea:40:5a:a6:2b是什么设备

lhf19900402, 2021-04-24:
> 查一下08:ea:40:5a:a6:2b是什么设备

一个萤石云的网络摄像头。

Goodman-vip, 2021-04-26:
估计被黑了