stupidloud/nanopi-openwrt #588: 恩山上有人测试打上USB3.0控制器驱动后不互换网也可以跑满千兆

lulanyu, 2021-04-26:
https://www.right.com.cn/forum/thread-4107714-1-1.html

JackJie-J, 2021-04-26:
z早就看到了，不确定1988打没打。

aimuch, 2021-04-27:
> z早就看到了，不确定1988打没打。

已经打上了`xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed`
查看系统内核日志：    
```
[    0.209975] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    0.210556] ehci-pci: EHCI PCI platform driver
[    0.211018] ehci-platform: EHCI generic platform driver
[    0.213891] ehci-platform ff5c0000.usb: EHCI Host Controller
[    0.214428] ehci-platform ff5c0000.usb: new USB bus registered, assigned bus number 1
[    0.215280] ehci-platform ff5c0000.usb: irq 29, io mem 0xff5c0000
[    0.229799] ehci-platform ff5c0000.usb: USB 2.0 started, EHCI 1.00
[    0.231096] hub 1-0:1.0: USB hub found
[    0.231478] hub 1-0:1.0: 1 port detected
[    0.232411] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    0.232997] ohci-platform: OHCI generic platform driver
[    0.233906] ohci-platform ff5d0000.usb: Generic Platform OHCI controller
[    0.234527] ohci-platform ff5d0000.usb: new USB bus registered, assigned bus number 2
[    0.235380] ohci-platform ff5d0000.usb: irq 30, io mem 0xff5d0000
[    0.298566] hub 2-0:1.0: USB hub found
[    0.298942] hub 2-0:1.0: 1 port detected
[    0.300516] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[    0.301034] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 3
[    0.301958] xhci-hcd xhci-hcd.0.auto: hcc params 0x0220fe64 hci version 0x110 quirks 0x0000000002010010
[    0.302869] xhci-hcd xhci-hcd.0.auto: irq 164, io mem 0xff600000
[    0.304420] hub 3-0:1.0: USB hub found
[    0.304804] hub 3-0:1.0: 1 port detected
[    0.305608] xhci-hcd xhci-hcd.0.auto: xHCI Host Controller
[    0.306168] xhci-hcd xhci-hcd.0.auto: new USB bus registered, assigned bus number 4
[    0.306861] xhci-hcd xhci-hcd.0.auto: Host supports USB 3.0 SuperSpeed
[    0.307533] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
[    0.308859] hub 4-0:1.0: USB hub found
```

qazesz1, 2021-04-27:
然而这个固件一直都没有互换跑满千兆的啊