stupidloud/nanopi-openwrt #594: 采用ext4文件系统，刷卡之后可使用分区工具对sd卡扩容根分区至最大？有没具体的教程？真小白不懂

SiupoLeung, 2021-05-01:
大佬们打扰了，关于采用ext4文件系统，刷卡之后可使用分区工具对sd卡扩容根分区至最大？有没具体的教程？真小白不懂
我试过在windows系统用dg 和傲梅分区助手，没做到，查过百度，是不是一定要linux环境才可以，或者ssh可以吗？希望大佬们有空指导一下，谢谢

xiaowonet, 2021-05-01:
卡刷完毕后，把卡插进路由然后通电，电脑ssh连接上路由，使用在线升级命令升级一次，就自动扩容了

SiupoLeung, 2021-05-01:
关于squashfs格式的固件，如作者大大能一起做了就解决上面的问题了，也希望能考虑，愿意为此付费，穷打工人，一百两百奶茶钱范围可以接受。本人设备R4S。                                                                                                                                                                            
                                                                                                                                                           来自本人opwrt从入门到放弃

SiupoLeung, 2021-05-01:
> 卡刷完毕后，把卡插进路由然后通电，电脑ssh连接上路由，使用在线升级命令升级一次，就自动扩容了

谢谢指导 我太难了

Monstar199803, 2021-05-01:
> 大佬们打扰了，关于采用ext4文件系统，刷卡之后可使用分区工具对sd卡扩容根分区至最大？有没具体的教程？真小白不懂
> 我试过在windows系统用dg 和傲梅分区助手，没做到，查过百度，是不是一定要linux环境才可以，或者ssh可以吗？希望大佬们有空指导一下，谢谢

https://mlapp.cn/1011.html 你可以在这里看看

i-never-left-you, 2021-05-02:
只要在线升级一次就可以了

JackJie-J, 2021-05-03:
你只需要在线升级一次，就自动扩展到你卡剩余最大存储了