stupidloud/nanopi-openwrt #599: 修改seed文件后生成固件失败

qazesz1, 2021-05-03:
我修改r2s.config.seed，然后在action里生成r2s固件，提示Process completed with exit code 126错误
修改后文件如下，只是注释掉一些行，没有添加新的
```
CONFIG_TARGET_rockchip=y
CONFIG_TARGET_rockchip_armv8=y
CONFIG_TARGET_rockchip_armv8_DEVICE_friendlyarm_nanopi-r2s=y
CONFIG_DEVEL=y
# CONFIG_CCACHE is not set
CONFIG_TARGET_OPTIONS=y
CONFIG_TARGET_OPTIMIZATION="-O3 -pipe -march=armv8-a+crypto+crc -mcpu=cortex-a53+crypto+crc -mtune=cortex-a53"
CONFIG_TARGET_ROOTFS_PARTSIZE=600
CONFIG_TARGET_ROOTFS_EXT4FS=y
# CONFIG_TARGET_ROOTFS_SQUASHFS is not set
# CONFIG_SIGNED_PACKAGES is not set
CONFIG_IB=y
CONFIG_IB_STANDALONE=y

CONFIG_PACKAGE_dnsmasq_full_dhcpv6=y
CONFIG_PACKAGE_ipv6helper=y
# CONFIG_PACKAGE_luci-app-adbyby-plus is not set
# CONFIG_PACKAGE_luci-app-adguardhome is not set
CONFIG_PACKAGE_luci-app-appfilter=y
CONFIG_PACKAGE_luci-app-argon-config=y
CONFIG_PACKAGE_luci-app-aria2=y
CONFIG_PACKAGE_luci-app-arpbind=y
# CONFIG_PACKAGE_luci-app-baidupcs-web is not set
CONFIG_PACKAGE_luci-app-cifsd=y
CONFIG_PACKAGE_luci-app-cpufreq=y
CONFIG_PACKAGE_luci-app-ddns=y
CONFIG_PACKAGE_ddnsto=y
CONFIG_PACKAGE_luci-app-ddnsto=y
CONFIG_PACKAGE_luci-app-diskman=y
CONFIG_PACKAGE_luci-app-dockerman=y
CONFIG_DOCKER_OPTIONAL_FEATURES=y
CONFIG_PACKAGE_luci-app-filetransfer=y
CONFIG_PACKAGE_luci-app-frpc=y
CONFIG_PACKAGE_luci-app-godproxy=y
# CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set
# CONFIG_PACKAGE_luci-app-jd-dailybonus is not set
CONFIG_PACKAGE_luci-app-mentohust=y
CONFIG_PACKAGE_luci-app-netdata=y
# CONFIG_PACKAGE_luci-app-nft-qos is not set
CONFIG_PACKAGE_luci-app-nlbwmon=y
# CONFIG_PACKAGE_luci-app-openclash is not set
# CONFIG_PACKAGE_luci-app-openvpn-server is not set
# CONFIG_PACKAGE_luci-app-passwall is not set
# CONFIG_PACKAGE_luci-app-qbittorrent is not set
# CONFIG_PACKAGE_luci-app-serverchan is not set
# CONFIG_PACKAGE_luci-app-smartdns is not set
CONFIG_PACKAGE_luci-app-softethervpn=y
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-statistics=y
# CONFIG_PACKAGE_luci-app-transmission is not set
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-app-turboacc=y
CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_shortcut-fe=y
CONFIG_PACKAGE_luci-app-udpxy=y
CONFIG_PACKAGE_luci-app-uhttpd=y
CONFIG_PACKAGE_luci-app-unblockmusic=y
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-usb-printer=y
# CONFIG_PACKAGE_luci-app-uugamebooster is not set
CONFIG_PACKAGE_luci-app-vlmcsd=y
CONFIG_PACKAGE_luci-app-vnstat=y
CONFIG_PACKAGE_luci-app-vsftpd=y
# CONFIG_PACKAGE_luci-app-vssr is not set
CONFIG_PACKAGE_luci-app-wol=y
# CONFIG_PACKAGE_luci-app-zerotier is not set
CONFIG_PACKAGE_luci-theme-argonv3=y
CONFIG_PACKAGE_luci-theme-edge=y
CONFIG_PACKAGE_luci-theme-material=y
CONFIG_PACKAGE_luci-theme-netgear=y
CONFIG_PACKAGE_luci-theme-rosy=y
CONFIG_PACKAGE_luci-theme-infinityfreedom=y
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_coremark=y
CONFIG_PACKAGE_coreutils-truncate=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_fdisk=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_iperf3=y
CONFIG_PACKAGE_ksmbd-utils=y
CONFIG_PACKAGE_losetup=y
CONFIG_PACKAGE_pv=y
CONFIG_PACKAGE_resize2fs=y
CONFIG_PACKAGE_sfdisk=y
CONFIG_PACKAGE_usb-modeswitch=y
CONFIG_PACKAGE_wget=y
# CONFIG_OPENSSL_ENGINE_BUILTIN_AFALG is not set
# CONFIG_OPENSSL_ENGINE_BUILTIN_DEVCRYPTO is not set
CONFIG_PACKAGE_collectd-mod-ping=y
CONFIG_PACKAGE_collectd-mod-thermal=y
CONFIG_PACKAGE_ddns-scripts_cloudflare.com-v4=y
CONFIG_PACKAGE_iptables-mod-conntrack-extra=y
CONFIG_PACKAGE_iptables-mod-ipopt=y
CONFIG_PACKAGE_iptables-mod-tproxy=y
CONFIG_PACKAGE_kmod-tcp-bbr=y
CONFIG_PACKAGE_kmod-ipt-offload=y
CONFIG_PACKAGE_kmod-fast-classifier=y
CONFIG_PACKAGE_kmod-pppol2tp=y
CONFIG_PACKAGE_kmod-zram=y

CONFIG_DROPBEAR_ECC=y
CONFIG_DROPBEAR_ECC_FULL=y

CONFIG_PACKAGE_kmod-drm-rockchip=y
CONFIG_PACKAGE_kmod-gpu-lima=y
CONFIG_PACKAGE_kmod-mt76x0u=y
CONFIG_PACKAGE_kmod-mt76x2u=y
CONFIG_PACKAGE_kmod-rtl8821cu=y
CONFIG_PACKAGE_kmod-rtl88x2bu=y
CONFIG_PACKAGE_iw=y
CONFIG_PACKAGE_iwinfo=y
CONFIG_PACKAGE_wpad-wolfssl=y
CONFIG_DRIVER_11AC_SUPPORT=y
CONFIG_DRIVER_11N_SUPPORT=y
CONFIG_DRIVER_11W_SUPPORT=y

# CONFIG_PACKAGE_luci-app-oled=y
```

作为对照，不修改就能成功

请问这是哪没做对呢

xiaowonet, 2021-05-03:
试试把自己注释的那些行全部删除，而不是仅仅注释掉

klever1988, 2021-05-03:
是的 把不要的东西整行删去 不要用注释或not set

qazesz1, 2021-05-04:
直接删掉没问题了，但是后面我想加一个 luci-app-wireguard，又失败了。这个包我在openwrt_buildinfo里面查过是有的，不能直接写到seed里吗？
```

CONFIG_PACKAGE_luci-app-wireguard=y

```

```
 * opkg_install_cmd: Cannot install package luci-app-wireguard.
Configuring luci-i18n-statistics-zh-cn.
make[2]: *** [package_install] Error 255
Configuring luci-i18n-zerotier-zh-cn.
make[1]: *** [_call_image] Error 2
```

qazesz1, 2021-05-04:
还有个问题，按照一楼的精简以后，从首页看到总可用内存只有700多M了，难道这些插件还会影响内存？