stupidloud/nanopi-openwrt #616: 一个网口的neo3，刷r2s固件后要怎样访问后台？

lyricusark, 2021-05-09:
neo3做旁路由用，卡刷固件后接入局域网，通过 http://immortalwrt/ 和主路由分配的IP地址，都无法进入网页管理界面。

baby195lxl, 2021-05-10:
> neo3做旁路由用，卡刷固件后接入局域网，通过 http://immortalwrt/ 和主路由分配的IP地址，都无法进入网页管理界面。

neo3和r2sde固件可以通用吗？

lyricusark, 2021-05-10:
> 
> 
> > neo3做旁路由用，卡刷固件后接入局域网，通过 http://immortalwrt/ 和主路由分配的IP地址，都无法进入网页管理界面。
> 
> neo3和r2sde固件可以通用吗？

去年minimal版本的时候是可以通用的，这版不知可否已调整兼容下。

baby195lxl, 2021-05-10:
> > > neo3做旁路由用，卡刷固件后接入局域网，通过 http://immortalwrt/ 和主路由分配的IP地址，都无法进入网页管理界面。
> > 
> > 
> > neo3和r2sde固件可以通用吗？
> 
> 去年minimal版本的时候是可以通用的，这版不知可否已调整兼容下。

如果可以的话，那neo3做旁路由还不错

lyricusark, 2021-05-10:
k大的版本就是没测通，现在我试过的这个neo3是可以用的https://github.com/biliwala/nanopi-openwrt/releases/tag/R2S-Minimal-2021-05-10

klever1988, 2021-05-11:
可以正常地给客户端dhcp到ip吗？

lyricusark, 2021-05-11:
> 
> 
> 可以正常地给客户端dhcp到ip吗？

设备只有一个网口，没加wifi。neo3接主路由可以获取到ip，电脑直接网线接neo3，电脑端IP获取有问题。

klever1988, 2021-05-11:
> > 可以正常地给客户端dhcp到ip吗？
> 
> 设备只有一个网口，没加wifi。neo3接主路由可以获取到ip，电脑直接网线接neo3，电脑端IP获取有问题。

接主路由的时候要把wan防火墙设置为允许入站
接电脑时那你要把网口切换为lan接口才行

lyricusark, 2021-05-11:
是不是卡刷固件后，neo3这个设备相当于只有一个wan口？请问那要怎样设置wan防火墙允许入站，或是改lan口呢，现在是登录不了网页管理界面，putty访问也被拒绝。

lyricusark, 2021-05-11:
可否请K大在编译时，预设wan口防火墙允许入站，这样单口路由就好操作了，谢谢大佬!

lyricusark, 2021-05-16:
> 
> 
> > > 可以正常地给客户端dhcp到ip吗？
> > 
> > 
> > 设备只有一个网口，没加wifi。neo3接主路由可以获取到ip，电脑直接网线接neo3，电脑端IP获取有问题。
> 
> 接主路由的时候要把wan防火墙设置为允许入站
> 接电脑时那你要把网口切换为lan接口才行

K大，我只有一个wan口，默认拒绝入站没法改设置啊。最后买了个COMFAST 811AC，临时用wifi接入改了设置解决了。

closed 2021-05-16 by lyricusark

al123xiaaaa, 2024-04-03:
> > > 可以正常地给客户端dhcp到ip吗？
> > 
> > 
> > 设备只有一个网口，没加wifi。neo3接主路由可以获取到ip，电脑直接网线接neo3，电脑端IP获取有问题。
> 
> 接主路由的时候要把wan防火墙设置为允许入站 接电脑时那你要把网口切换为lan接口才行

感谢，解决我的问题了