stupidloud/nanopi-openwrt #629: 为啥我的R2S用端内更新的办法不能更新呢？

chwl66, 2021-05-15:
我使用的命令是这个`wget -qO- https://raw.githubusercontent.com/klever1988/nanopi-openwrt/master/scripts/autoupdate.sh | ver=-with-docker sh`，当我按了回车以后直接跳到下一行 什么反应都没有。
[image]
如图所示！
有没有大佬碰到这种问题的，这个应该怎么去解决呢？

sujialong123, 2021-05-15:
最近两天一直是这种情况，刚刚试了一下又有新花样了
[image]

chwl66, 2021-05-15:
> 最近两天一直是这种情况，刚刚试了一下又有新花样了
> [image]

你的应该是命令少了个w 应该是wget

sujialong123, 2021-05-15:
wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-with-docker sh

sujialong123, 2021-05-21:
wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-with-docker sh

命令补齐了结果还是一样，问题也没人解决。。。

974656746, 2021-05-23:
可能网络屏蔽了github域名，你开个V2或者s*s*r爬墙就好

chwl66, 2021-05-23:
感谢，尝试用扶墙方法，可以更新，但是更新的时候CPU占用很高，导致R2S卡死，最后远程重启，升级也失败了，不敢再更新了，怕出问题！