stupidloud/nanopi-openwrt #630: 提交个IPsec VPN的问题

IZANAI, 2021-05-17:
klever大大您好！

问题是这样的：

首先我在外网经由ipsec回到内网以后，发现几个情况：
1、无法ping通除R2S以外的其他内网设备IP，也无法上网；
2、微信可以发文字，接收缩略图，但是无法发图片无法查看原图。

就排查了半天，然后发现vpn0其实是个摆设，还有个ipsec0，这个也不算bug吧，毕竟小丑是我自己

于是乎把ipsec0加入区域
[image]

然后访问内网ip就ok了，当我认为大功告成的时候……还是无法访问网页

DNS无法解析，我已经在VPN设置里面设置的R2S的LAN地址作为网关，也在IPsecVPN的设置页面设置R2S的LAN地址作为网关
[image]

然后情况就是可以走通内网但是走不通internet，应该是DNS又有问题
[image]

但是内网设备是可以走通internet的。

就很迷……

盼解答

lucifer662607004, 2021-06-11:
我也遇到同样问题，请问
"于是乎把ipsec0加入区域"
这一步是怎么做的？

kqkq, 2021-08-02:
网络 - 防火墙 - 通信规则 - 新建转发规则

名称随便起，源区域VPN，目标区域lan

点添加并编辑，全部默认值保存&应用

wolonggl, 2022-04-30:
我测试存在同样的问题，tcpdump抓包，已接受到DNS请求，但是dns请求未返回给ipsec客户端，看样子是DNS返回的请求被拦截h了，因为直接ip访问是正常的。
官方openwrt 21.02版本是无该问题的。