stupidloud/nanopi-openwrt #646: 请问有同学遇到R2S重启，但无法启动，sys灯常亮

silverjava, 2021-05-24:
我发现如果在web界面重启系统，或者使用reboot定时重启，都会出现R2S无响应，sys灯常亮，无论LAN口还是WAN灯都不会亮。

是我等的时间不够长，还是本身重启功能有问题？

我使用的是slim版本

deartonyz, 2021-05-24:
我也遇到了，换了充电头然后启动了！ 之前一直用去年的Minimal版本一直都启动很顺利的

silverjava, 2021-05-25:
你换电源好了吗？

deartonyz, 2021-05-26:
好了哦

1592554701, 2021-06-30:
我也是有这个问题， 今天打算换TF卡和电源 试试看还会不会卡重启了

yolyzhu, 2021-08-17:
只要用系统内重启，网卡就不亮，拔电源才能修复……

coolangcn, 2021-08-24:
只要用系统内重启，网卡就不亮，拔电源才能修复……我也是如此 WAN口不亮 LAN口亮 拔电源才能修复
请问是什么原因呢？电源问题？

coolangcn, 2021-08-24:
> 我也是有这个问题， 今天打算换TF卡和电源 试试看还会不会卡重启了

请问解决没有

yolyzhu, 2021-08-24:
@coolangcn 亲测换电源或者换卡都能修复，只要换对了就行。卡要那种速率慢的，不喜欢慢速卡就换个电源也可能可以。

Freikys, 2021-08-24:
@deartonyz 你换的啥电源啊？我用的钉钉打卡机的5V4A电源。

coolangcn, 2021-08-24:
> @coolangcn 亲测换电源或者换卡都能修复，只要换对了就行。卡要那种速率慢的，不喜欢慢速卡就换个电源也可能可以。
谢谢，我的是5V2A的华为的手机充电器 我回去换一下试试

klever1988, 2021-08-24:
我自己也发现了，主要还是卡的原因。
貌似我朗科卡一直没毛病，夏科卡一直无法重启

1592554701, 2021-08-25:
我换了京东上那个很便宜的科朗的卡，电源没换也管用了 刷的QiuSimons  已经可以正常重启，并且稳定运行一个多月了。同时也买了三星的Endurance PRO卡来测试感觉还是科朗的卡兼容稳定点  ，不知道是不是特例，另外，我刷的不是EXT4分区的包。

hitsmaxft, 2022-11-06:
从tty里可以看到文本停止输出，但不会重新引导需要断电，我这里一直有这个问题

elecompu1234, 2025-11-16:
经常碰到这种问题，这种DIY的硬件真的折腾人，感觉是厂家硬件设计生产的BUG。毕竟只是一个生产开发板的小作坊。根本不会做产品级的相关设计研究，折磨死人。在考虑换平台。

hitsmaxft, 2025-11-16:
> 经常碰到这种问题，这种DIY的硬件真的折腾人，感觉是厂家硬件设计生产的BUG。毕竟只是一个生产开发板的小作坊。根本不会做产品级的相关设计研究，折磨死人。在考虑换平台。

其实严格讲是openwrt的兼容性问题