stupidloud/nanopi-openwrt #667: R2S-Minimal-2020-12-14如何升级至最新版本

waldoxhm, 2021-06-02:
请问R2S-Minimal-2020-12-14如何在线升级至最新版本，可以使用“wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-with-docker sh”脚本直接升级到有docker的最新版吗？买了R2S就用了最后一个minimal版本稳定使用至今，十分感谢Klever1988大佬的付出！

WilsonXu0512, 2021-06-07:
> 请问R2S-Minimal-2020-12-14如何在线升级至最新版本，可以使用“wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | ver=-with-docker sh”脚本直接升级到有docker的最新版吗？买了R2S就用了最后一个minimal版本稳定使用至今，十分感谢Klever1988大佬的付出！

劝你别升，Minimal是基于Friendly源码改的，稳定性和速度比现在的版本好，现在的版本基于openwrt源码改的，有可能出现奇怪的问题还找不到原因。比如我只能在Minimal下跑满足，现在这个怎么都不行。我已经退回用Minimal了了

waldoxhm, 2021-06-09:
> 劝你别升，Minimal是基于Friendly源码改的，稳定性和速度比现在的版本好，现在的版本基于openwrt源码改的，有可能出现奇怪的问题还找不到原因。比如我只能在Minimal下跑满足，现在这个怎么都不行。我已经退回用Minimal了了

感谢指点！现在minimal各项功能都满足，你这一说劝退了，不升了

closed 2021-06-09 by waldoxhm