stupidloud/nanopi-openwrt #682: 自编译成功了

pp875598763, 2021-06-16:
留了passwall adguardhome docker smartdns 网络唤醒 跟kms 
去除不需要的服务 界面看着舒服多了

docker怎么玩的 有人教吗

13311031900, 2021-06-17:
怎么编译的教教吧

wspeed0711, 2021-06-17:
呵呵，我用的slim版，docker一直没折腾成功。

pp875598763, 2021-06-17:
> 怎么编译的教教吧

就fork 然后改r2s对应的文件 然后action里面选build 填入r2s 点run就可以了

heningcai, 2021-06-22:
请问一下，那个 可用软件的列表 在哪里查看  具体路径怎么走 [https://user-images.githubusercontent.com/56048681/114531174-3beafb80-9c7e-11eb-8bcc-b098c3b1cee8.png](url)

pp875598763, 2021-06-22:
> 请问一下，那个 可用软件的列表 在哪里查看 具体路径怎么走
[image]
点这里进去可以看到

中文对照表https://github.com/klever1988/nanopi-openwrt/blob/master/luci_app_manual.md

heningcai, 2021-06-26:
[image]
这个不知道在哪里

pp875598763, 2021-06-26:
> [image]
> 这个不知道在哪里
> 话说自己编译的有docker吗

会自动编译带docker的版本 你自己编译的时候加个docker管理端

pp875598763, 2021-06-26:
> [image]
> 这个不知道在哪里
> 话说自己编译的有docker吗

[image]

carlosmind, 2021-06-26:
请问自己编译完成之后从哪里下载啊，可以烦请您截个图教我一下吗，感谢！

pp875598763, 2021-06-26:
> 请问自己编译完成之后从哪里下载啊，可以烦请您截个图教我一下吗，感谢！

[image]

carlosmind, 2021-06-27:
> > 请问自己编译完成之后从哪里下载啊，可以烦请您截个图教我一下吗，感谢！
> 
> [image]

太感谢了