stupidloud/nanopi-openwrt #689: 不定时断网

Leakless, 2021-06-20:
症状：开pt下载，满速下载大概10分钟左右开始不定时断网，等几秒会好

kernel log:

[10274.721978] ------------[ cut here ]------------
[10274.722404] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
[10274.723006] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:481 dev_watchdog+0x2f4/0x300
[10274.723725] Modules linked in: xt_FULLCONENAT pppoe ppp_async mt76x0u mt76x0_common l2tp_ppp rtl88x2bu rtl8821cu rtl8812au pppox ppp_mppe ppp_generic mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76 mac80211 lzo ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY vgem usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr slhc rc_core r8152 phy_rockchip_inno_hdmi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp
[10274.723814]  nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo_decompress lzo_compress iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables exfat dw_hdmi_cec crc_ccitt compat asn1_decoder xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel tunnel4 ip_tunnel vfat fat nls_utf8 nls_cp437 zram zsmalloc shortcut_fe_ipv6 shortcut_fe sha1_generic ecb arc4 crypto_acompress fuse uas
[10274.731445]  gpio_button_hotplug
[10274.739217] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.124 #0
[10274.739739] Hardware name: FriendlyElec NanoPi R2S (DT)
[10274.740195] pstate: 20000005 (nzCv daif -PAN -UAO)
[10274.740616] pc : dev_watchdog+0x2f4/0x300
[10274.740966] lr : dev_watchdog+0x2f4/0x300
[10274.741316] sp : ffff800010013da0
[10274.741606] x29: ffff800010013da0 x28: ffff00003c67aa80 
[10274.742069] x27: 0000000000000004 x26: 0000000000000140 
[10274.742532] x25: 00000000ffffffff x24: 0000000000000002 
[10274.742996] x23: ffff00003c3b945c x22: ffff00003c3b9000 
[10274.743460] x21: ffff00003c3b9480 x20: ffff800010cc6000 
[10274.743923] x19: 0000000000000000 x18: 0000000000000000 
[10274.744386] x17: 0000000000000000 x16: 0000000000000000 
[10274.744849] x15: 0000000000000000 x14: ffff800010d57fc2 
[10274.745312] x13: 0000000000000000 x12: ffff800010d57000 
[10274.745776] x11: ffff800010cdb000 x10: 0000000000000010 
[10274.746239] x9 : 0000000000000000 x8 : 2030206575657571 
[10274.746703] x7 : 2074696d736e6172 x6 : ffff00003f595818 
[10274.747167] x5 : ffff00003f595818 x4 : 0000000000000004 
[10274.747631] x3 : 0000000000000000 x2 : 0000000000000004 
[10274.748094] x1 : ffff00003d8da700 x0 : 0000000000000039 
[10274.748558] Call trace:
[10274.748776]  dev_watchdog+0x2f4/0x300
[10274.749100]  call_timer_fn.isra.35+0x20/0x78
[10274.749473]  run_timer_softirq+0x378/0x388
[10274.749832]  __do_softirq+0x124/0x260
[10274.750154]  irq_exit+0xb8/0xc8
[10274.750431]  __handle_domain_irq+0x64/0xb8
[10274.750788]  gic_handle_irq+0x50/0xa8
[10274.751109]  el1_irq+0xb8/0x180
[10274.751387]  cpuidle_enter_state+0x84/0x358
[10274.751752]  cpuidle_enter+0x34/0x48
[10274.752066]  call_cpuidle+0x1c/0x40
[10274.752372]  do_idle+0x1c0/0x218
[10274.752654]  cpu_startup_entry+0x20/0x78
[10274.753000]  secondary_start_kernel+0x16c/0x180
[10274.753395] ---[ end trace 242d9f25036d4b27 ]---
[10274.753855] r8152 4-1:1.0 eth1: Tx timeout
[10274.756787] r8152 4-1:1.0 eth1: Tx status -2
[10274.757192] r8152 4-1:1.0 eth1: Tx status -2
[10274.757588] r8152 4-1:1.0 eth1: Tx status -2
[10274.758010] r8152 4-1:1.0 eth1: Tx status -2
[10276.889721] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[10276.922011] r8152 4-1:1.0 eth1: Invalid ether addr 00:00:00:00:00:00
[10276.922576] r8152 4-1:1.0 eth1: Random ether addr 8e:01:24:51:04:ae
[10276.924596] r8152 4-1:1.0 eth1: Promiscuous mode enabled

system log:

Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.721978] ------------[ cut here ]------------
Sun Jun 20 15:23:32 2021 kern.info kernel: [10274.722404] NETDEV WATCHDOG: eth1 (r8152): transmit queue 0 timed out
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.723006] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:481 dev_watchdog+0x2f4/0x300
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.723725] Modules linked in: xt_FULLCONENAT pppoe ppp_async mt76x0u mt76x0_common l2tp_ppp rtl88x2bu rtl8821cu rtl8812au pppox ppp_mppe ppp_generic mt76x2u mt76x2_common mt76x02_usb mt76x02_lib mt76_usb mt76 mac80211 lzo ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY vgem usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_fsm ts_bm tcp_bbr slhc rc_core r8152 phy_rockchip_inno_hdmi nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.723814]  nf_conntrack_sip nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan lzo_decompress lzo_compress iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip_tables exfat dw_hdmi_cec crc_ccitt compat asn1_decoder xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel tunnel4 ip_tunnel vfat fat nls_utf8 nls_cp437 zram zsmalloc shortcut_fe_ipv6 shortcut_fe sha1_generic ecb arc4 crypto_acompress fuse uas
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.731445]  gpio_button_hotplug
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.739217] CPU: 2 PID: 0 Comm: swapper/2 Not tainted 5.4.124 #0
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.739739] Hardware name: FriendlyElec NanoPi R2S (DT)
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.740195] pstate: 20000005 (nzCv daif -PAN -UAO)
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.740616] pc : dev_watchdog+0x2f4/0x300
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.740966] lr : dev_watchdog+0x2f4/0x300
[... 16 lines cut ...]
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.748558] Call trace:
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.748776]  dev_watchdog+0x2f4/0x300
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.749100]  call_timer_fn.isra.35+0x20/0x78
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.749473]  run_timer_softirq+0x378/0x388
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.749832]  __do_softirq+0x124/0x260
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.750154]  irq_exit+0xb8/0xc8
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.750431]  __handle_domain_irq+0x64/0xb8
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.750788]  gic_handle_irq+0x50/0xa8
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.751109]  el1_irq+0xb8/0x180
[... cut ...]
Sun Jun 20 15:23:32 2021 kern.warn kernel: [10274.758010] r8152 4-1:1.0 eth1: Tx status -2
Sun Jun 20 15:23:35 2021 kern.info kernel: [10276.889721] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
Sun Jun 20 15:23:35 2021 kern.err kernel: [10276.922011] r8152 4-1:1.0 eth1: Invalid ether addr 00:00:00:00:00:00
Sun Jun 20 15:23:35 2021 kern.info kernel: [10276.922576] r8152 4-1:1.0 eth1: Random ether addr 8e:01:24:51:04:ae
Sun Jun 20 15:23:35 2021 kern.notice kernel: [10276.924596] r8152 4-1:1.0 eth1: Promiscuous mode enabled****

klever1988, 2021-06-20:
在r2s上开pt吗？

Leakless, 2021-06-22:
下挂群晖，跑PT，下载速度大了就跪了
https://github.com/fanck0605/openwrt-nanopi-r2s/issues/10  好像和这个一样

klever1988, 2021-06-23:
下载的时候观察一下负载数字

Leakless, 2021-06-24:
负载观察过，没问题，都正常

michael-james0207, 2021-06-29:
> 下载的时候观察一下负载数字

我也会同样的情况，即使不是下载的时候，正常使用网络看剧，或者浏览网页也会出现断网重拨，原本以为是光猫的问题，联系运营商更换了一个全新的千兆还是有同样的问题

finch-ift, 2021-11-17:
我也遇到了 会无预兆断网30秒左右 然后就恢复了

quietcoolwu, 2022-03-31:
> > 下载的时候观察一下负载数字
> 
> 我也会同样的情况，即使不是下载的时候，正常使用网络看剧，或者浏览网页也会出现断网重拨，原本以为是光猫的问题，联系运营商更换了一个全新的千兆还是有同样的问题

请问这个是怎么解决的 现在还是碰到这个问题

michael-james0207, 2022-03-31:
重新刷固件，从最新的一个版本回刷，一个一个测试，测试稳定了就不要再折腾了，我最后测试出来跟用的clash版本有关联，所以你可以先测试纯净版的固件

发自我的iPhone