stupidloud/nanopi-openwrt #693: R4S slim版，luci-app-upnp软件包安装失败

zshleon, 2021-06-23:
slim版固件，
版本：ImmortalWrt 18.06-SNAPSHOT r10249-f90a84833 / LuCI openwrt-18.06-k5.4 branch (git-21.170.50917-05a4f95)

在自带的软件源里查找安装luci-app-upnp失败，错误提示如下：

 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.124-1-825a83d49f2511dc8ae89ac005238245) for kmod-nf-conntrack-netlink
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-nf-conntrack-netlink found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-upnp:
 * 	kernel (= 5.4.124-1-825a83d49f2511dc8ae89ac005238245)
 * opkg_install_cmd: Cannot install package luci-app-upnp.

klever1988, 2021-06-23:
emmm,  记得是几号的吗

zshleon, 2021-06-23:
最新的，昨天的，下午跑了下在线更新的脚本，提示没有发现更新。

klever1988, 2021-06-26:
应该是21号 然后我下了21号的版本 检查local_feed的packages索引
[image]

看了一下内核模块的依赖，摘要确认是802d08cfd53642661468bb82f3a23d78，
你可以检查/local_feed/Packages用文本编辑器打开搜索Package: kmod-nf-conntrack-netlink，看看是825a83d49f2511dc8ae89ac005238245还是802d08cfd53642661468bb82f3a23d78