stupidloud/nanopi-openwrt #703: 求增加编译WireGuard

hwangjunjie, 2021-07-03:
先感谢klever1988大大的固件！！！

希望使用WireGuard，但是尝试安装失败，尝试自己编译也失败，求帮忙支持，再次感谢

Collected errors:
Configuring libyaml.
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-blake2s.
Configuring ruby-psych.
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-chacha20.
Configuring ruby-yaml.
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-poly1305.
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-chacha20poly1305.
Configuring coreutils-nohup.
Configuring luci-app-openclash.
 * opkg_install_cmd: Cannot install package kmod-crypto-kpp.
Configuring kmod-ipt-conntrack-extra.
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-curve25519.
 * opkg_install_cmd: Cannot install package kmod-wireguard.
 * opkg_install_cmd: Cannot install package luci-proto-wireguardr.
 * opkg_install_cmd: Cannot install package luci-i18n-wireguard-zh-cn.
 * opkg_install_cmd: Cannot install package wireguard-tools.
 * opkg_install_cmd: Cannot install package luci-app-wireguard.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_image] Error 2
make: *** [image] Error 2

closed 2021-07-04 by hwangjunjie

messense, 2021-07-25:
我选择降级到 2021-07-11 的版本了：https://github.com/klever1988/nanopi-openwrt/releases/tag/2021-07-11

klever1988, 2021-07-25:
WG没有预装可以从slim软件中心安装啊
https://github.com/klever1988/nanopi-openwrt/blob/master/extra_packages.seed

messense, 2021-07-25:
怎么安装？非 slim 版本固件。

klever1988, 2021-07-25:
非slim版本用的是官方软件源，有wireguard但是依赖的kmod可能因为内核签名不一致无法安装。
只能github actions生成，添加CONFIG_PACKAGE_luci-app-wireguard=y到r2s.config.seed，然后从extra_package.seed删除CONFIG_PACKAGE_luci-app-wireguard=m（因为可能会覆盖前面的=y配置）