stupidloud/nanopi-openwrt #704: SMB网络存储，复制大文件时出现“无效的文件句柄”

GinChen0525, 2021-07-03:
背景：R4S的usb口外接希捷4t的2.5寸机械硬盘，使用绿联的usb3.0转sata线
症状：复制大文件时出现：“无效的文件句柄”，导致复制失败

自己研究步骤：
1.进后台修改了文件句柄限制：echo 100000 > /proc/sys/fs/file-max
2.编辑/etc/sysctl.conf，文件最后加入fs.file-max=100000
3.将ulimit 加入到/etc/profile 文件底部：echo ulimit -n 65535 >>/etc/profile

发现稳定了许多，但依然没有解决问题！

继续解决：
1.禁止autosuspend：echo -1 > /sys/module/usbcore/parameters/autosuspend
2.卸载了kmod-usb-storage-uas软件包（参考：https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1584557）

最终问题依然没有解决。。。
贴上一部分系统日志与内核日志：

> Sun Jul  4 02:39:27 2021 kern.err kernel: [18200.462146] usb 8-1: cmd cmplt err -71
Sun Jul  4 02:39:27 2021 kern.err kernel: [18200.462547] usb 8-1: cmd cmplt err -71
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.530629] scsi host2: uas_pre_reset: timed out
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.531391] xhci-hcd xhci-hcd.1.auto: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.532273] xhci-hcd xhci-hcd.1.auto: WARN Set TR Deq Ptr cmd failed due to incorrect slot or ep state.
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.533131] sd 2:0:0:0: [sdc] tag#5 uas_zap_pending 0 uas-tag 1 inflight: CMD
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.533777] sd 2:0:0:0: [sdc] tag#5 CDB: opcode=0x8a 8a 00 00 00 00 00 00 f1 e3 00 00 00 04 00 00 00
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.534599] sd 2:0:0:0: [sdc] tag#4 uas_zap_pending 0 uas-tag 2 inflight: CMD
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.535242] sd 2:0:0:0: [sdc] tag#4 CDB: opcode=0x8a 8a 00 00 00 00 00 00 f1 df 00 00 00 04 00 00 00
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.536052] sd 2:0:0:0: [sdc] tag#6 uas_zap_pending 0 uas-tag 3 inflight: CMD
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.536693] sd 2:0:0:0: [sdc] tag#6 CDB: opcode=0x8a 8a 00 00 00 00 00 00 f1 e7 00 00 00 03 80 00 00
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.538300] sd 2:0:0:0: [sdc] tag#5 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=0x00
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.539051] sd 2:0:0:0: [sdc] tag#5 CDB: opcode=0x8a 8a 00 00 00 00 00 00 f1 e3 00 00 00 04 00 00 00
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.539860] blk_update_request: I/O error, dev sdc, sector 15852288 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.540906] sd 2:0:0:0: [sdc] tag#4 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=0x00
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.541632] sd 2:0:0:0: [sdc] tag#4 CDB: opcode=0x8a 8a 00 00 00 00 00 00 f1 df 00 00 00 04 00 00 00
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.542438] blk_update_request: I/O error, dev sdc, sector 15851264 op 0x1:(WRITE) flags 0x0 phys_seg 128 prio class 0
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.543421] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 891682816 size 917504 starting block 1981504)
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.544657] buffer_io_error: 6875 callbacks suppressed
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.544665] Buffer I/O error on device sdc1, logical block 1981024
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.545676] Buffer I/O error on device sdc1, logical block 1981025
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.546226] Buffer I/O error on device sdc1, logical block 1981026
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.546806] Buffer I/O error on device sdc1, logical block 1981027
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.547356] Buffer I/O error on device sdc1, logical block 1981028
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.547904] Buffer I/O error on device sdc1, logical block 1981029
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.548452] Buffer I/O error on device sdc1, logical block 1981030
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.548999] Buffer I/O error on device sdc1, logical block 1981031
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.549546] Buffer I/O error on device sdc1, logical block 1981032
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.550095] Buffer I/O error on device sdc1, logical block 1981033
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.551276] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 892600320 size 90112 starting block 1981526)
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.552548] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 892690432 size 20480 starting block 1981531)
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.553771] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 892710912 size 4096 starting block 1981532)
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.555010] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 892715008 size 4096 starting block 1981533)
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.556213] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 892719104 size 4096 starting block 1981534)
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.557416] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 892723200 size 8192 starting block 1981536)
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.558693] sd 2:0:0:0: [sdc] tag#6 UNKNOWN(0x2003) Result: hostbyte=0x01 driverbyte=0x00
Sun Jul  4 02:39:32 2021 kern.info kernel: [18205.559419] sd 2:0:0:0: [sdc] tag#6 CDB: opcode=0x8a 8a 00 00 00 00 00 00 f1 e7 00 00 00 03 80 00 00
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.560225] blk_update_request: I/O error, dev sdc, sector 15853312 op 0x1:(WRITE) flags 0x0 phys_seg 112 prio class 0
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.561168] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 892731392 size 917504 starting block 1981760)
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.563039] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 893648896 size 65536 starting block 1981776)
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.564348] blk_update_request: I/O error, dev sdc, sector 15854208 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.565333] blk_update_request: I/O error, dev sdc, sector 15855232 op 0x1:(WRITE) flags 0x4000 phys_seg 112 prio class 0
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.566298] EXT4-fs warning (device sdc1): ext4_end_bio:315: I/O error 10 writing to inode 239730699 (offset 893714432 size 983040 starting block 1982016)
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.568225] blk_update_request: I/O error, dev sdc, sector 15856128 op 0x1:(WRITE) flags 0x0 phys_seg 25 prio class 0
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.569248] blk_update_request: I/O error, dev sdc, sector 15856328 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.570244] blk_update_request: I/O error, dev sdc, sector 15857352 op 0x1:(WRITE) flags 0x0 phys_seg 119 prio class 0
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.570260] blk_update_request: I/O error, dev sdc, sector 15858304 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.572025] blk_update_request: I/O error, dev sdc, sector 15859328 op 0x1:(WRITE) flags 0x0 phys_seg 128 prio class 0
Sun Jul  4 02:39:32 2021 kern.warn kernel: [18205.580513] JBD2: Detected IO errors while flushing file data on sdc1-8
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.581165] Aborting journal on device sdc1-8.
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.581626] Buffer I/O error on dev sdc1, logical block 488374272, lost sync page write
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.582387] Buffer I/O error on dev sdc1, logical block 0, lost sync page write
Sun Jul  4 02:39:32 2021 kern.err kernel: [18205.583051] EXT4-fs (sdc1): I/O error while writing superblock
[... 134 lines cut ...]
[18260.318451] blk_update_request: I/O error, dev sdc, sector 2063612160 op 0x1:(WRITE) flags 0x4000 phys_seg 128 prio class 0
[18260.320015] blk_update_request: I/O error, dev sdc, sector 2063613184 op 0x1:(WRITE) flags 0x0 phys_seg 128 prio class 0
[... cut ...]
[18261.093373] sd 2:0:0:0: [sdc] Write Protect is off
[18261.093799] sd 2:0:0:0: [sdc] Mode Sense: 43 00 00 00
[18261.093985] sd 2:0:0:0: [sdc] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[18261.095082] sd 2:0:0:0: [sdc] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
[18261.140292]  sdc: sdc1
[18261.142003] sd 2:0:0:0: [sdc] Attached SCSI disk
[18262.240636] EXT4-fs (sdc1): recovery complete
[18262.274546] EXT4-fs (sdc1): mounted filesystem with ordered data mode. Opts:

GinChen0525, 2021-07-03:
想了想，是不是问题出在了：**cmd cmplt err -71**这个上面？

可是如何解决呢？

GinChen0525, 2021-07-03:
更有趣的是，闲置不动时系统日志会做如下记录：

> Sun Jul  4 02:41:29 2021 kern.info kernel: [18323.164165] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:41:52 2021 kern.info kernel: [18346.149200] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:42:15 2021 kern.info kernel: [18369.088280] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:42:21 2021 kern.info kernel: [18374.936289] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:42:24 2021 kern.info kernel: [18377.484365] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:42:45 2021 kern.info kernel: [18399.296483] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:42:56 2021 kern.info kernel: [18409.780528] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:42:56 2021 kern.info kernel: [18410.228600] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:43:19 2021 kern.info kernel: [18432.620608] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:44:15 2021 kern.info kernel: [18489.040950] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:44:17 2021 kern.info kernel: [18490.741118] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:44:30 2021 kern.info kernel: [18503.773274] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:44:39 2021 kern.info kernel: [18512.629158] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:45:36 2021 kern.info kernel: [18570.337559] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:45:42 2021 kern.info kernel: [18575.865738] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:51:16 2021 kern.info kernel: [18910.271857] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:51:17 2021 kern.info kernel: [18911.372196] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:51:24 2021 kern.info kernel: [18917.548116] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:51:29 2021 kern.info kernel: [18923.287893] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:53:24 2021 kern.info kernel: [19037.564787] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:53:52 2021 kern.info kernel: [19065.925083] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:53:57 2021 kern.info kernel: [19070.860902] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:53:58 2021 kern.info kernel: [19071.652856] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:54:14 2021 kern.info kernel: [19087.764998] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:59:28 2021 kern.info kernel: [19401.703055] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 02:59:30 2021 kern.info kernel: [19404.259202] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:00:36 2021 kern.info kernel: [19469.671499] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:00:38 2021 kern.info kernel: [19471.643539] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:00:39 2021 kern.info kernel: [19472.971495] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:00:41 2021 kern.info kernel: [19474.507441] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:00:42 2021 kern.info kernel: [19476.043437] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:05:09 2021 kern.info kernel: [19742.565095] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:05:13 2021 kern.info kernel: [19747.057219] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:05:22 2021 kern.info kernel: [19756.125265] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:06:03 2021 kern.info kernel: [19796.865752] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:06:07 2021 kern.info kernel: [19800.649565] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd
Sun Jul  4 03:06:11 2021 kern.info kernel: [19805.065455] usb 8-1: reset SuperSpeed Gen 1 USB device number 82 using xhci-hcd

这又是怎么回事？

klever1988, 2021-07-03:
会不会是盘本身有问题了，看看smart信息

GinChen0525, 2021-07-04:
> 会不会是盘本身有问题了，看看smart信息

[image]

应该不会，京东旗舰店刚买的硬盘

GinChen0525, 2021-07-04:
今天看了许多国外的论坛，发现linux貌似对uas支持不够友好，部分用户通过将设备强行驱动为usb-storage并解决问题。

> 参考：https://www.yumao.name/read/openwrt-mtk-usb3-uas/
（本人尝试后依然没有成功更改驱动模式 - - 

@klever1988 咱有办法解决不？

ShunYea, 2021-07-04:
> 
> 
> > 会不会是盘本身有问题了，看看smart信息
> 
> [image]
> 
> 应该不会，京东旗舰店刚买的硬盘

你硬盘用的NTFS格式吗？

klever1988, 2021-07-04:
给转接线换个固件版本试试 我记得以前有些usb-sata硬盘柜子在linux上运行就不太好 只能换低版本固件解决
或者如果是硬盘休眠的问题 试试曲线救国 写个脚本定时读取硬盘上的数据 避免休眠

GinChen0525, 2021-07-05:
> > > 会不会是盘本身有问题了，看看smart信息
> > 
> > 

> > 应该不会，京东旗舰店刚买的硬盘
> 
> 你硬盘用的NTFS格式吗？

当然不是，使用diskgenius改的ext4，nfts识别不到呀

GinChen0525, 2021-07-05:
> 给转接线换个固件版本试试 我记得以前有些usb-sata硬盘柜子在linux上运行就不太好 只能换低版本固件解决
> 或者如果是硬盘休眠的问题 试试曲线救国 写个脚本定时读取硬盘上的数据 避免休眠

好的，我先尝试更换不持支usap的usb3.0转sata线看看。（绿联默认支持usap）
应该不可能是硬盘休眠，因为每次都是在拷贝大文件时出现“无效文件句柄”提示

GinChen0525, 2021-07-07:
更新：
更换了不支持usap的转接线，**成功了！！！**
果然是usp驱动的锅
[image]

经测试，传输大文件很稳定，不会出现“无效文件句柄”提示，表明修改默认文件句柄数值无用

**但还剩下一个小问题：**

> kern.info kernel: [255439.720636] usb 6-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
这个提示依然会出现，但频率不高（会短暂中断硬盘的读写）

暂时找不到比较好的解决办法，如果其他伙伴有的话欢迎交流

HiGarfield, 2022-02-08:
Try to backport the following fix to the linux kernel.
https://lore.kernel.org/linux-usb/20191009044647.24536-1-heinzelmann.david@gmail.com/