stupidloud/nanopi-openwrt #705: 请教大神的固件怎么加上多拨功能呢？

ShunYea, 2021-07-04:
其他挺好的，而且大神的固件比较精简，就是少了多拨和负载均衡，看到可以自己编译我就fork了一份，修改了组件的config.seed，结果编译失败，是我组件修改有问题吗？请教一下，或是能在你的固件上帮加上这个功能，万分感谢。

klever1988, 2021-07-04:
你编译成功了啊   但是github actions触发编译那边要输入r4s

ShunYea, 2021-07-05:
> 
> 
> 你编译成功了啊 但是github actions触发编译那边要输入r4s

我触发那里是r2s成功了，但是改成r4s就报错了，然后被我删除了。请教一下多拨是要加入哪几个组件呢？你看我改的r2s的那个对吗？加了两个地方。

ShunYea, 2021-07-05:
> 
> 
> 你编译成功了啊 但是github actions触发编译那边要输入r4s

还有你说的“添加自己所需的软件，可用软件的列表可以在github actions构件输出处获取”，一直找不到这个输出处是哪个地方。

klever1988, 2021-07-05:
> > 你编译成功了啊 但是github actions触发编译那边要输入r4s
> 
> 我触发那里是r2s成功了，但是改成r4s就报错了，然后被我删除了。请教一下多拨是要加入哪几个组件呢？你看我改的r2s的那个对吗？加了两个地方。

你别删，我可以看日志呢

klever1988, 2021-07-05:
> > 你编译成功了啊 但是github actions触发编译那边要输入r4s
> 
> 还有你说的“添加自己所需的软件，可用软件的列表可以在github actions构件输出处获取”，一直找不到这个输出处是哪个地方。

[image]

ShunYea, 2021-07-05:
> 
> 
> > > 你编译成功了啊 但是github actions触发编译那边要输入r4s
> > 
> > 
> > 我触发那里是r2s成功了，但是改成r4s就报错了，然后被我删除了。请教一下多拨是要加入哪几个组件呢？你看我改的r2s的那个对吗？加了两个地方。
> 
> 你别删，我可以看日志呢

我没删除了，你看下报错了。编译的r4s，刚又试了下r2s还是成功，就是r4s不行

ShunYea, 2021-07-05:
> 
> 
> > > 你编译成功了啊 但是github actions触发编译那边要输入r4s
> > 
> > 
> > 还有你说的“添加自己所需的软件，可用软件的列表可以在github actions构件输出处获取”，一直找不到这个输出处是哪个地方。
> 
> [image]

为啥我还是看不到……没有下面的artifacts的窗口。

klever1988, 2021-07-05:
https://github.com/klever1988/nanopi-openwrt/actions/runs/996895268

ShunYea, 2021-07-05:
> 
> 
> https://github.com/klever1988/nanopi-openwrt/actions/runs/996895268

谢谢，原来是要在你的这里查，不是在我fork过去那边查。你能帮看看这个r4s报错的问题吗，又编译了一次，我没删除错误的

klever1988, 2021-07-06:
现在可以了 你再试试

ShunYea, 2021-07-07:
> 
> 
> 现在可以了 你再试试

编译了r2s和r4s的版本，都已经用上了，谢谢。