stupidloud/nanopi-openwrt #709: R4S使用docker固件时会出现降速

NathanDai, 2021-07-07:
r4s刷了doxker固件 正常开机 但是第二天网速会下降500m的宽带变成了100M 重启路由器可以恢复正常

<sub>Sent from <a href="https://apps.apple.com/cn/app/id1314212521">PPHub</a></sub>

OoNanGeOo, 2021-07-10:
你也有这个问题？我也是这个问题，R2S

klever1988, 2021-07-10:
用ethtool查看是哪个网口掉到了100M模式，然后换网线

sdmkjz, 2021-07-11:
> 用ethtool查看是哪个网口掉到了100M模式，然后换网线

我也发现了存在这个问题 带docker固件的 我是r2s 刚开机时测速能满千兆 过段时间就下降到了400m左右

NathanDai, 2021-07-12:
> > 用ethtool查看是哪个网口掉到了100M模式，然后换网线
> 
> 我也发现了存在这个问题 带docker固件的 我是r2s 刚开机时测速能满千兆 过段时间就下降到了400m左右

我换了个固件目前还是正常

sdmkjz, 2021-07-12:
> 
> 
> > > 用ethtool查看是哪个网口掉到了100M模式，然后换网线
> > 
> > 
> > 我也发现了存在这个问题 带docker固件的 我是r2s 刚开机时测速能满千兆 过段时间就下降到了400m左右
> 
> 我换了个固件目前还是正常

目前用的哪家的 对docker有需求

smilence86, 2021-07-12:
我r2s刷了7月固件也是，隔壁[https://github.com/bleach1991/lede](https://github.com/bleach1991/lede) 也是降速，现在退回到6月1号继续观察。

Mattosx, 2021-07-16:
r2s 也出现了，应该出现好几个月了。现象是使用几天以后，测速的时候 ksoftirqd 进程占用很高，速度上不去，经过ss 可以跑200m的变成40m，重启可以解决。

smilence86, 2021-07-17:
我也是，最初刷 bleach1991/lede 的r2s高大全固件发现降速，不算梯子，第二天开始基础宽带从200M降到40M，cpu负载达到3点几，进入ssh排查，发现ksoftirqd/1、ksoftirqd/2占用cpu资源较多，还不能结束，重启恢复正常。

过来刷klever1988的r2s docker固件，退回到6月1号，同样降速。

sdmkjz, 2021-07-17:
> 
> 
> 我也是，最初刷 bleach1991/lede 的r2s高大全固件发现降速，不算梯子，第二天开始基础宽带从200M降到40M，cpu负载达到3点几，进入ssh排查，发现ksoftirqd/1、ksoftirqd/2占用cpu资源较多，还不能结束进程，重启恢复正常。
> 
> 过来刷klever1988的r2s docker固件，退回到6月1号，同样降速。

“有个文章说网络风暴或包循环转发会造成这种问题，引文受机器cpu和网卡物理速度的限制一个系统的包转发能力是有极限的，而风暴或循环最容易到达这种极限。” 看样有可能是底层内核源码的问题

NMDaDa, 2021-07-20:
> > > > 用ethtool查看是哪个网口掉到了100M模式，然后换网线
> > > 
> > > 
> > > 我也发现了存在这个问题 带docker固件的 我是r2s 刚开机时测速能满千兆 过段时间就下降到了400m左右
> > 
> > 
> > 我换了个固件目前还是正常
> 
> 目前用的哪家的 对docker有需求

求问目前用的哪家，降速降得烦

smilence86, 2021-07-20:
我刷了6月1号非docker版r2s固件，还是降速，但进程ksoftirqd/2没有消耗cpu资源了。