stupidloud/nanopi-openwrt #715: 固件降速问题

sdmkjz, 2021-07-11:
R2S-docker版本 实测开机用一段时间后会降速 通过命令查看端口握手速率没有问题 在Issues中看到有朋友说是openclash的问题 但是通过测试 感觉跟openclash并没有关系 目前在用的插件有 clash 网络唤醒 adg ddns docker 京东签到

sdmkjz, 2021-07-11:
另外建议作者网络唤醒换另一个版本的 这个版本一但电脑关机后 就识别不到MAC和IP 需要手动输入 之前我用过plus那个版本的 可以保存IP和MAC

klever1988, 2021-07-11:
哪个plus版本？

sdmkjz, 2021-07-11:
> 哪个plus版本？

CONFIG_PACKAGE_luci-app-services-wolplus=y
这个我添加到seed中编译过 没成功 这个plus版本在404的固件当中有

NMDaDa, 2021-07-14:
docker版本有降速问题，刷无docker4天宽带未降速。但有新问题，运行一天后出国速度降低，不知道是什么问题

klever1988, 2021-07-14:
> > 哪个plus版本？
> 
> CONFIG_PACKAGE_luci-app-services-wolplus=y
> 这个我添加到seed中编译过 没成功 这个plus版本在404的固件当中有
> 另外降速问题还烦请作者排查下 我现在还没排查出哪里的问题 下午重启后到现在还没降速

整进去了

sdmkjz, 2021-07-16:
> 
> 
> > > 哪个plus版本？
> > 
> > 
> > CONFIG_PACKAGE_luci-app-services-wolplus=y
> > 这个我添加到seed中编译过 没成功 这个plus版本在404的固件当中有
> > 另外降速问题还烦请作者排查下 我现在还没排查出哪里的问题 下午重启后到现在还没降速
> 
> 整进去了

收到 不过新版本的Docker遇到问题了 已发新的iessues

muronghan, 2021-07-16:
7月初的R2S的无Docker版本，没开OpenClash，开了很少的插件包括HelloWorld，降速，从500Mbps降到200Mbps

smilence86, 2021-07-22:
我发现降速是由ipv6引起的，最近刷了好几个r2s固件，开ssrp、agh，基本上第二天就会降速，同时2个叫ksoftirqd/1、ksoftirqd/2的进程占用大量cpu资源，引发负载升高。

跟是否docker没关系
跟sfe是否开启没关系

关闭wan、lan的ipv6，运行2天了没降速。

sdmkjz, 2021-07-22:
> 
> 
> 我发现降速是由ipv6引起的，最近刷了好几个r2s固件，开ssrp、agh，基本上第二天就会降速，同时2个叫ksoftirqd/1、ksoftirqd/2的进程占用大量cpu资源，引发负载升高。
> 
> 跟是否docker没关系
> 跟sfe是否开启没关系
> 
> 关闭wan、lan的ipv6，运行2天了没降速。

我现在换了silm版本 就装了几个插件 但IPV6正常用 五天了没有降速

smilence86, 2021-07-22:
那我也换slim版算了，r2s做主路由只负责拨号，pve虚拟x86 op做旁路由解决其他需求。

这个r2s买来一年多问题太多，前几个月因为供电问题经常死机，最近几个月又没找到稳定可靠固件。

sdmkjz, 2021-07-22:
> 
> 
> 那我也换slim版算了，r2s做主路由只负责拨号，pve虚拟x86 op做旁路由解决其他需求。
> 
> 这个r2s买来一年多问题太多，前几个月因为供电问题经常死机，最近几个月又没找到稳定可靠固件。

R2S挑卡挑电源 这俩搞好了实际上用起来还是比较稳定的 固件都差不多 我原先用404固件 后来因为docker需求 换了1988大佬的 现在silm版本我也没装太多东西 adg也不用了 ssrp也换成clash了

NMDaDa, 2021-07-26:
确实，更新新无docker后还会出现降速。现已安装slim版本，运行3天，宽带及出国均为降速，持续关注中

smilence86, 2021-07-26:
我刷19号slim版2天了没降速，之前导出了6月1号无docker版的配置，导入恢复后slim版就自动出现那些插件了，也是神奇。