stupidloud/nanopi-openwrt #716: tar的版本是不是有什么问题

hwdef, 2021-07-12:


klever1988, 2021-07-12:
busybox的tar不是原版的tar，参数可能有不同

hwdef, 2021-07-13:
有什么办法可以装原版的tar吗

klever1988, 2021-07-13:
opkg update
opkg install tar

hwdef, 2021-07-13:
```shell
root@ImmortalWrt:~# opkg update
Downloading file:///local_feed/Packages.gz
Updated list of available packages in /var/opkg-lists/local
root@ImmortalWrt:~# opkg install tar
Unknown package 'tar'.
Collected errors:
 * opkg_install_cmd: Cannot install package tar.
 ```
没有这个包

klever1988, 2021-07-13:
添加一下这个源

src/gz immortalwrt_packages https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages

hwdef, 2021-07-14:
```shell
root@ImmortalWrt:~# cat  /etc/opkg/distfeeds.conf
src/gz local file:///local_feed
#src/gz immortalwrt_core https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages
#src/gz immortalwrt_base https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base
src/gz immortalwrt_packages https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages
#src/gz immortalwrt_luci https://mirrors.tencent.com/lede/releases/18.06.9/packages/aarch64_generic/luci
#src/gz immortalwrt_routing https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing
#src/gz immortalwrt_telephony https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony
root@ImmortalWrt:~# opkg update
Downloading file:///local_feed/Packages.gz
Updated list of available packages in /var/opkg-lists/local
Downloading https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages/Packages.gz
Updated list of available packages in /var/opkg-lists/immortalwrt_packages
root@ImmortalWrt:~# opkg install tar 
Unknown package 'tar'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for tar found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package tar.
```

好像cpu的架构不对。

hwdef, 2021-07-14:
```shell
root@ImmortalWrt:~# cat /proc/cpuinfo 
processor       : 0
model name      : ARMv8 Processor rev 4 (v8l)
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv8 Processor rev 4 (v8l)
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv8 Processor rev 4 (v8l)
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv8 Processor rev 4 (v8l)
BogoMIPS        : 48.00
Features        : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

r2s nanopi

klever1988, 2021-07-14:
src/gz immortalwrt_base https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base

base源也启用一下吧

hwdef, 2021-07-15:
可以了，谢谢

closed 2021-07-15 by hwdef