stupidloud/nanopi-openwrt #718: R2s好像只有2个核心工作？

445746695, 2021-07-14:
“统计”和“netdata”中均显示核心“1”和“2”负载50%左右，“0”和“3”基本空载，是否正常现象？是什么原因？如何解决？谢谢。

klever1988, 2021-07-14:
主要是软中断占用，一般是网卡
immortalwrt为了优化性能绑定了网卡软中断到指定核心上面，不过我也不知道作者怎么分配的
所以这个情况可以说是某个网卡的接收或者发出有大量调用，和你的openwrt设置也有关系
你可以取消绑定，系统就会自己分配，图表看起来会好看点，不过会对性能有所影响

445746695, 2021-07-16:
因为内网挂了小盒子，应该是eth0和eth1都有较高占用吧？或者说，nat负载高？
那这样的话我觉得应该4个核心使用率不会相差太多吧。还是不太理解这个“2核围观”的情况。4核同时使用对比目前2核工作的状况，还要降低性能？

445746695, 2021-07-16:
root@ImmortalWrt:~# cat /proc/interrupts
           CPU0       CPU1       CPU2       CPU3
  3:   65787596   81616711   94022249   65007786     GICv2  30 Level     arch_ti                                                                                        mer
  6:         13          0          0          0     GICv2  32 Level     ff1f000                                                                                        0.dmac
  7:          0          0          0          0     GICv2  33 Level     ff1f000                                                                                        0.dmac
  8:          0          0          0          0     GICv2 132 Level     arm-pmu
  9:          0          0          0          0     GICv2 133 Level     arm-pmu
 10:          0          0          0          0     GICv2 134 Level     arm-pmu
 11:          0          0          0          0     GICv2 135 Level     arm-pmu
 12:          1          0          0          0     GICv2  89 Level     ttyS2
 13:          0          0          0          0     GICv2  68 Level     ff15000                                                                                        0.i2c
 14:    9695739          0          0          0     GICv2  69 Level     ff16000                                                                                        0.i2c
 16:          0          0          0          0     GICv2  90 Level     rockchi                                                                                        p_thermal
 17:          0          0          0          0     GICv2 122 Level     gp
 18:          0          0          0          0     GICv2 119 Level     gpmmu
 19:          0          0          0          0     GICv2 125 Level     pp_bcas                                                                                        t
 20:          0          0          0          0     GICv2 120 Level     pp0
 21:          0          0          0          0     GICv2 121 Level     ppmmu0
 22:          0          0          0          0     GICv2 123 Level     pp1
 23:          0          0          0          0     GICv2 124 Level     ppmmu1
 25:          0          0          0          0     GICv2  43 Level     ff35080                                                                                        0.iommu
 26:    1218724          0          0          0     GICv2  44 Level     dw-mci
 27:         10 1373620812          0          0     GICv2  56 Level     eth0
 29:          0          0          0          0     GICv2  48 Level     ehci_hc                                                                                        d:usb1
 30:          0          0          0          0     GICv2  49 Level     ohci_hc                                                                                        d:usb2
 35:          0          0          0          0  rockchip_gpio_irq   0 Edge                                                                                              keys
 91:          0          0          0          0  rockchip_gpio_irq  24 Level                                                                                             rk805
163:          0          0          0          0     GICv2  94 Level     rockchi                                                                                        p_usb2phy
164:         23          0 3554692463          0     GICv2  99 Level     xhci-hc                                                                                        d:usb3
170:          0          0          0          0     rk805   5 Edge      RTC ala                                                                                        rm
IPI0: 114828207   46090932   22123869  126944094       Rescheduling interrupts
IPI1:       849        878        649        645       Function call interrupts
IPI2:         0          0          0          0       CPU stop interrupts
IPI3:         0          0          0          0       CPU stop (for crash dump)                                                                                         interrupts
IPI4:  17957791      48636      57137   16048738       Timer broadcast interrupt                                                                                        s
IPI5:    576161     984669    1090926     579567       IRQ work interrupts
IPI6:         0          0          0          0       CPU wake-up interrupts
Err:          0
root@ImmortalWrt:~# cat /proc/softirqs
                    CPU0       CPU1       CPU2       CPU3
          HI:          1          1          0          1
       TIMER:   61940860   64428374   84198711   59377226
      NET_TX:    2278695 1452490578    7684976    2270721
      NET_RX:      87464 2082954043 2673209527      89214
       BLOCK:      18340       2041       1719       4091
    IRQ_POLL:          0          0          0          0
     TASKLET:    6372461 1486946512   14662813    5178883
       SCHED:   42961855   42533412   46204943   37424997
     HRTIMER:      11615      28558       7552      11485
         RCU:   41078388   50175456   51415121   38739477