stupidloud/nanopi-openwrt #722: R2S-docker

sdmkjz, 2021-07-16:
最新版本openwrt 启动docker守护程序提示 Please input the SOCKET PATH of docker daemon! 都是默认配置 刚更新完启动就有这个提示 重启系统也一样

sdmkjz, 2021-07-16:
使用终端创建容器提示如下错误 docker: Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?.

sdmkjz, 2021-07-16:
通过实测 该问题是因为系统中/var/run/下不存在docker.sock文件

sdmkjz, 2021-07-16:
应该是因为缺少docker的引擎 请作者排查下编译过程是否落下了

klever1988, 2021-07-16:
估计新版dockerman插件还不完善，我用slim+luci-app-docker能用

sdmkjz, 2021-07-17:
> 估计新版dockerman插件还不完善，我用slim+luci-app-docker能用

大概是这样了 目前来看是缺少docker-ce引擎或者引擎未启动

hans362, 2021-07-18:
我这也是同样的情况，R2S Docker 版本在最近一次更新后 Dockerman 就报错了，但是实际上 Docker Daemon 还在，只是 `/etc/init.d` 下对应的 `dockerd` 服务不知为何消失了，我的临时解决方法是强制重装一遍 `dockerd`
```
opkg update
opkg install dockerd --force-reinstall
service dockerd restart
```
这样服务就回来了

sdmkjz, 2021-07-18:
> 我这也是同样的情况，R2S Docker 版本在最近一次更新后 Dockerman 就报错了，但是实际上 Docker Daemon 还在，只是 `/etc/init.d` 下对应的 `dockerd` 服务不知为何消失了，我的临时解决方法是强制重装一遍 `dockerd`
> ```
> opkg update
> opkg install dockerd --force-reinstall
> service dockerd restart
> ```
> 这样服务就回来了

对 我当时也是想重装 无奈不知道引擎重装的命令 最后换了silm版本🌚🌚