stupidloud/nanopi-openwrt #725: 请问大哥们如何挂载这个地方的容量

gekkjoe, 2021-07-17:
[image]
我想把上面的空间 
挂载到下面的空间
[image]

之前的版本自动挂载 7月16更新之后自动挂载就不行 自己上网找了些命令 最后把系统搞坏了

gekkjoe, 2021-07-17:
[image]
已经全部勾选了

klever1988, 2021-07-18:
把p3这个分区删除 然后在线升级一次就扩容了

gekkjoe, 2021-07-18:
> 把p3这个分区删除 然后在线升级一次就扩容了

好的大哥 请问在线升级一次 是什么意思?
[image]
是这个意思吗

smilence86, 2021-07-18:
我是参考这篇教程扩容：[https://seahi.me/263.html](https://seahi.me/263.html)，亲测有效。

另外，大佬能不能出个扩容脚本，只有扩容不升级最新版，我刷了最新版有问题，退回到历史版本继续观察。

前些天刷了最近几个月docker版r2s固件都会降速，昨晚刷了非docker版继续观察。

klever1988, 2021-07-18:
> > 把p3这个分区删除 然后在线升级一次就扩容了
> 
> 好的大哥 请问在线升级一次 是什么意思?
> [image]
> 是这个意思吗

把卡拔下来插到电脑，然后用diskgenius之类的分区工具删除p3分区，然后用
https://github.com/klever1988/nanopi-openwrt#%E7%BB%88%E7%AB%AF%E5%86%85%E5%9C%A8%E7%BA%BF%E5%8D%87%E7%BA%A7%E6%96%B9%E6%B3%95
升级

klever1988, 2021-07-18:
> 我是参考这篇教程扩容：https://seahi.me/263.html，亲测有效。
> 
> 另外，大佬能不能出个扩容脚本，只有扩容不升级最新版，我刷了最新版有问题，退回到历史版本继续观察。
> 
> 前些天刷了最近几个月docker版r2s固件都会降速，昨晚刷了非docker版继续观察。

只扩容不升级的话，最方便的方式就是拔卡接到linux(op也可以)的主机，用cfdisk扩容p2分区信息，然后用resize2fs扩容文件系统
教程介绍其实是在线“更换”根分区到新建的大分区，并不是扩容，非overlay没办法在线扩容，只能拔卡离线操作

gekkjoe, 2021-07-18:
> 只扩容不升级的话，最方便的方式就是拔卡接到linux(op也可以)的主机，用cfdisk扩容p2分区信息，然后用resize2fs扩容文件系统
> 教程介绍其实是在线“更换”根分区到新建的大分区，并不是扩容，非overlay没办法在线扩容，只能拔卡离线操作

> 我是参考这篇教程扩容：https://seahi.me/263.html，亲测有效。
> 
> 另外，大佬能不能出个扩容脚本，只有扩容不升级最新版，我刷了最新版有问题，退回到历史版本继续观察。
> 
> 前些天刷了最近几个月docker版r2s固件都会降速，昨晚刷了非docker版继续观察。

感谢两位大爷抽空解答

gekkjoe, 2021-07-19:
[image]
[image]
[image]

大哥 刷完固件再在线升级也没法扩容

> 只扩容不升级的话，最方便的方式就是拔卡接到linux(op也可以)的主机，用cfdisk扩容p2分区信息，然后用resize2fs扩容文件系统
> 教程介绍其实是在线“更换”根分区到新建的大分区，并不是扩容，非overlay没办法在线扩容，只能拔卡离线操作