stupidloud/nanopi-openwrt #733: 大佬们求救SmartDNS遇到一些问题  也搞不清楚开启了没有

gekkjoe, 2021-07-20:
[image]
[image]
[image]
[image]
[image]
请问如果是把重定向关了
[image]
然后
[image]
按道理应该是可行 为啥不行的?

gekkjoe, 2021-07-20:
[image]
这样设置是不是100%所有请求都走到5335呢?跟53再无关系 对不对

yuwansiniana, 2021-07-20:
要不试试adguard home 设置起来也挺简单的。基本的dns功能和广告过滤也都能实现。

gekkjoe, 2021-07-20:
> 要不试试adguard home 设置起来也挺简单的。基本的dns功能和广告过滤也都能实现。

老哥 请问自动切换节点你有什么好推荐 我感觉SSR+的自动切换不是太灵活

yuwansiniana, 2021-07-20:
passwall用着还不错

gekkjoe, 2021-07-20:
> passwall用着还不错

但是小弟不会安装 而且还有好多依赖包  55555

gekkjoe, 2021-07-20:
> passwall用着还不错

不会装

jzh33, 2021-07-22:
恕我直言，smartdns想法和理念非常好，但是并无实际意义，本地dns一般来说肯定是运营商提供的解析最快，科学上网本身已经有成熟的dns解决方案了，所以这个smartdns设置起来很麻烦实际提升感受很有限，别喷我，我只是实话实说。