stupidloud/nanopi-openwrt #742: 在线升级没成功过2月份版本

bobobomb, 2021-07-26:
--2021-07-26 22:02:26--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-07-26/r2s.img.gz
Resolving ghproxy.com... 198.18.5.80
Connecting to ghproxy.com|198.18.5.80|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 133182497 (127M) [application/octet-stream]
Saving to: 'STDOUT'

-                                     100%[======================================================================>] 127.01M  6.16MB/s    in 26s     

2021-07-26 22:02:55 (4.85 MB/s) - written to stdout [133182497/133182497]

gzip: stdin: decompression OK, trailing garbage ignored
今天的固件还没更新，尝试下载昨天的固件
--2021-07-26 22:02:55--  https://ghproxy.com/https://github.com/klever1988/nanopi-openwrt/releases/download/2021-07-25/r2s.img.gz
Resolving ghproxy.com... 198.18.5.80
Connecting to ghproxy.com|198.18.5.80|:443... connected.
HTTP request sent, awaiting response... 404 NOT FOUND
2021-07-26 22:02:58 ERROR 404: NOT FOUND.

gzip: stdin: unexpected end of file
没找到最新的固件，脚本退出

klever1988, 2021-07-26:
你的tmp分区容量剩余多大

gekkjoe, 2021-07-26:
> 你的tmp分区容量剩余多大

会判断剩余容量??

derekling, 2021-07-27:
可以后台上传更新吗？