stupidloud/nanopi-openwrt #746: K大佬麻烦处理一下在线升级总是中断

gekkjoe, 2021-07-27:
wget -qO- https://github.com/klever1988/nanopi-openwrt/raw/master/scripts/autoupdate.sh | sh

[image]

gekkjoe, 2021-07-27:
[image]

klever1988, 2021-07-28:
https://github.com/klever1988/nanopi-openwrt/blob/master/scripts/autoupdate.sh#L70

直接下载这个脚本，编辑sleep 5为sleep 10或者更大的数字，
然后直接执行编辑后的autoupdate.sh试试

`sh autoupdate.sh`

derekling, 2021-07-28:
> https://github.com/klever1988/nanopi-openwrt/blob/master/scripts/autoupdate.sh#L70
> 
> 直接下载这个脚本，编辑sleep 5为sleep 10或者更大的数字，
> 然后直接执行编辑后的autoupdate.sh试试
> 
> `sh autoupdate.sh`

在哪里执行

klever1988, 2021-07-28:
> > https://github.com/klever1988/nanopi-openwrt/blob/master/scripts/autoupdate.sh#L70
> > 直接下载这个脚本，编辑sleep 5为sleep 10或者更大的数字，
> > 然后直接执行编辑后的autoupdate.sh试试
> > `sh autoupdate.sh`
> 
> 在哪里执行

ssh终端里啊