stupidloud/nanopi-openwrt #759: R2S升级

adamzhou111, 2021-08-03:
各位大佬，R2S可以直接在线升级吗？升级过程中会断网，或者升级完配置都还在吗？
[image]

klever1988, 2021-08-03:
你这个版本好旧了，用另一张卡试吧

adamzhou111, 2021-08-03:
> 你这个版本好旧了，要是担心，用另一张卡试吧

换张SD卡，使用终端在线升级吗？配置需要全部手工设置吗？

klever1988, 2021-08-03:
> > 你这个版本好旧了，要是担心，用另一张卡试吧
> 
> 换张SD卡，使用终端在线升级吗？配置需要全部手工设置吗？

你可以换张卡用balenaEtcher刷入，然后用旧系统导出配置再导入新系统。有什么异常的话，返回使用旧系统那张卡不就好了

adamzhou111, 2021-08-03:
> > > 你这个版本好旧了，要是担心，用另一张卡试吧
> > 
> > 
> > 换张SD卡，使用终端在线升级吗？配置需要全部手工设置吗？
> 
> 你可以换张卡用balenaEtcher刷入，然后用旧系统导出配置再导入新系统。有什么异常的话，返回使用旧系统那张卡不就好了

[image]
这个导出，是指导出配置吗？

klever1988, 2021-08-03:
> > > > 你这个版本好旧了，要是担心，用另一张卡试吧
> > > 
> > > 
> > > 换张SD卡，使用终端在线升级吗？配置需要全部手工设置吗？
> > 
> > 
> > 你可以换张卡用balenaEtcher刷入，然后用旧系统导出配置再导入新系统。有什么异常的话，返回使用旧系统那张卡不就好了
> 
> [image]
> 这个导出，是指导出配置吗？

是的

adamzhou111, 2021-08-03:
> > > > > 你这个版本好旧了，要是担心，用另一张卡试吧
> > > > 
> > > > 
> > > > 换张SD卡，使用终端在线升级吗？配置需要全部手工设置吗？
> > > 
> > > 
> > > 你可以换张卡用balenaEtcher刷入，然后用旧系统导出配置再导入新系统。有什么异常的话，返回使用旧系统那张卡不就好了
> > 
> > 
> > [image]
> > 这个导出，是指导出配置吗？
> 
> 是的

好的，非常感谢