stupidloud/nanopi-openwrt #797: 使用 Ksmbd 共享后，网络里面进不去 提示 0x80070053

bingkina, 2021-08-19:
使用 Ksmbd 共享后，网络里面进不去 提示 0x80070053
在 win10 上已经开启了 SMB 服务了
[image]
[image]
[image]

klever1988, 2021-08-20:
我也一样 不知道什么问题

bingkina, 2021-08-20:
> 我也一样 不知道什么问题

然后直接在地址栏输入 \\192.168....路由器地址，就又变成这个问题了 https://github.com/klever1988/nanopi-openwrt/issues/479
好惨

klever1988, 2021-08-20:
https://www.jianshu.com/p/e0820155bfdf

试一下，我可以

bingkina, 2021-08-21:
> https://www.jianshu.com/p/e0820155bfdf
> 
> 试一下，我可以

修改后可以进了，但是还是这问题。。。登不进去 https://github.com/klever1988/nanopi-openwrt/issues/479
[image]
[image]

Neustradamus, 2021-09-16:
To follow this bug.