stupidloud/nanopi-openwrt #798: 编译失败了 。merge_packages.sh这个文件是增加插件库的吗？

Seventtang, 2021-08-20:
想增加一个这个插件 luci-app-openvpn   增加后 编译R2S固件失败 。在猜想是不是下载问题还是插件库没这个呢。

merge_packages.sh这个文件是增加插件库的吗？  为什么会失败呢

Seventtang, 2021-08-20:
失败提示：Error: Process completed with exit code 2.

Seventtang, 2021-08-20:
这是日志地址：[https://pipelines.actions.githubusercontent.com/vXTiawiDRnv0NM6Kap0TYkVZrmpLcENlNzE8cAsPgiS2Y1hJWU/_apis/pipelines/1/runs/23/signedlogcontent/7?urlExpires=2021-08-20T13%3A27%3A56.5588084Z&urlSigningMethod=HMACV1&urlSignature=W7Pxo%2BrFpdcJazCyK3IM89WoJuFWR4bU9FuheMAsWPY%3D](url)

Seventtang, 2021-08-20:
https://pipelines.actions.githubusercontent.com/vXTiawiDRnv0NM6Kap0TYkVZrmpLcENlNzE8cAsPgiS2Y1hJWU/_apis/pipelines/1/runs/23/signedlogcontent/7?urlExpires=2021-08-20T13%3A27%3A56.5588084Z&urlSigningMethod=HMACV1&urlSignature=W7Pxo%2BrFpdcJazCyK3IM89WoJuFWR4bU9FuheMAsWPY%3D

Seventtang, 2021-08-20:
日志好多文字怎么复制过来呢。。

klever1988, 2021-08-21:
之前没有预编译这个插件，所以你生成的时候会出错。 现在有了，你再试试