stupidloud/nanopi-openwrt #801: R2S在线升级以后软件源不见了

chwl66, 2021-08-21:
[image]
软件源变成这样了，
[image]
这个里面的参数我也不知道替换成啥！

klever1988, 2021-08-21:
这是slim版本，内置的本地软件源，配置就长这样的。
你刷新一下再搜索软件试试。

chwl66, 2021-08-21:
> 这是slim版本，内置的本地软件源，配置就长这样的。
> 你刷新一下再搜索软件试试。

我刷的是with-docker版，升级也是，升级完就不见了！刷新搜索到也是不能安装。
[image]
使用opkg update命令没反应的。没办法安装。

klever1988, 2021-08-21:
你是从slim升级到docker版的

> > 这是slim版本，内置的本地软件源，配置就长这样的。
> > 你刷新一下再搜索软件试试。
> 
> 我刷的是with-docker版，升级也是，升级完就不见了！刷新搜索到也是不能安装。
> [image]
> 使用opkg update命令没反应的。没办法安装。

复制一下完整的输出

chwl66, 2021-08-21:
#722 我是按照这个里面的命令强装dockerd的 结果和软件包里面安装是一样的。
[image]
卡刷的时候用的17号的包，带docker的
[image]

klever1988, 2021-08-21:
你刷的是docker的版本，怎么源配置会是slim的呢。。

你把源配置更新一下吧
src/gz immortalwrt_core https://mirrors.tencent.com/lede/snapshots/targets/rockchip/armv8/packages
src/gz immortalwrt_base https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/base
src/gz immortalwrt_packages https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/packages
src/gz immortalwrt_luci https://mirrors.tencent.com/lede/releases/18.06.9/packages/aarch64_generic/luci
src/gz immortalwrt_routing https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/routing
src/gz immortalwrt_telephony https://mirrors.tencent.com/lede/snapshots/packages/aarch64_generic/telephony