stupidloud/nanopi-openwrt #808: R2S安装openvpn-openssl报错

minkaka, 2021-08-24:
2021年8月22日发布的 r2s.img.gz，在软件中安装openvpn-openssl，报错以下信息。

`Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Package openvpn-openssl (2.5.3-2) installed in root is up to date.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 5.4.142-1-97621c2479d8ecf1e0a4b0bf2eaca93f) for kmod-tun`

minkaka, 2021-08-24:
添加openvon配置时，页面弹出

```

/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/openvpn'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/openvpn.lua:64: bad argument #4 to 'set' (string expected, got boolean)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

```

allinu, 2022-04-09:
为啥我现在的界面就是没有上传文件的这个按钮也不知大咋操作

[image]

有好心人帮帮我吗

Maxine114, 2023-04-25:
> 为啥我现在的界面就是没有上传文件的这个按钮也不知大咋操作
> 
> [image]
> 
> 有好心人帮帮我吗

请问你现在解决了吗？我也是这个问题

Maxine114, 2023-04-25:
> 添加openvon配置时，页面弹出
>
> ```
>
> /usr/lib/lua/luci/dispatcher.lua:469: Failed to execute cbi dispatcher target for entry '/admin/services/openvpn'.
> The called action terminated with an exception: [...]

大哥解决了吗？我也是这个问题