stupidloud/nanopi-openwrt #810: 大神能不能加进openvpn客户端啊

minkaka, 2021-08-24:
r2s.img.gz
r2s-slim.img.gz
这2个镜像刷好后默认没有，需要在软件界面安装，安装时就报错，安装好后也无法增加openvpn配置文件

calkchu, 2021-08-24:
同样有需求，而且能否增加openvpn的多客户端对单服务器的配置

nigelgo, 2021-08-25:
同样需求，openssl安装报错

nigelgo, 2021-08-28:
> 这里是相关的参考https://openwrt.org/docs/guide-user/services/vpn/openvpn/client-luci

你弄好了吗

minkaka, 2021-08-28:
全弄好了，然后手贱重刷了一次固件又配置不起来了，头昏。--
Yours,

Bigcar

minkaka, 2021-08-28:
> 你弄好了吗
终于搞好了，世界清静了。

minkaka, 2021-08-28:
> > 这里是相关的参考https://openwrt.org/docs/guide-user/services/vpn/openvpn/client-luci
> 
> 你弄好了吗

终于重新搞好了，世界清静了。

klever1988, 2021-08-28:
> > > 这里是相关的参考https://openwrt.org/docs/guide-user/services/vpn/openvpn/client-luci
> > 
> > 
> > 你弄好了吗
> 
> 终于重新搞好了，世界清静了。
> [image]

重新加入了openvpn客户端，也是snapshot快照的最新版本，你不试试吗

minkaka, 2021-08-31:
> > > > 这里是相关的参考https://openwrt.org/docs/guide-user/services/vpn/openvpn/client-luci
> > > 
> > > 
> > > 你弄好了吗
> > 
> > 
> > 终于重新搞好了，世界清静了。
> > [image]
> 
> 重新加入了openvpn客户端，也是snapshot快照的最新版本，你不试试吗

谢谢大佬出手，今天就试试，试完来汇报。

allinu, 2022-04-09:
emmm, 为啥我的R2S安装完之后没有文件上传的部分

[image]

而且现在这个框，我也不知道填写什么

hkboy2000, 2022-05-02:
我也是一樣，没有文件上传的部分...也不知道填写什么

elliot-bia, 2022-09-19:
> > > > 这里是相关的参考https://openwrt.org/docs/guide-user/services/vpn/openvpn/client-luci
> > > 
> > > 
> > > 你弄好了吗
> > 
> > 
> > 终于重新搞好了，世界清静了。
> > [image]
> 
> 重新加入了openvpn客户端，也是snapshot快照的最新版本，你不试试吗

> 配置

大佬，这个是slim版本上有的吗？我这边怎么没找到
[image]
[image]