stupidloud/nanopi-openwrt #811: R2S的zerotier

iamfuhui, 2021-08-24:
更新了r2s的系统后，发现没有了zerotier。
请问是永久取消这个插件了么？

klever1988, 2021-08-24:
看着需求比较少 就从胖版里拿出来了 slim版还是有预置的

iamfuhui, 2021-08-24:
> 看着需求比较少 就从胖版里拿出来了 slim版还是有预置的

难过的是，我自己安装了好像还不能用。

iamfuhui, 2021-08-24:
好的，还是希望你可以把zerotier加入原版。我先关了这个issue。

klever1988, 2021-08-24:
> > 看着需求比较少 就从胖版里拿出来了 slim版还是有预置的
> 
> 难过的是，我自己安装了好像还不能用。

你怎么装的？slim从软件源安装还是第三方下载的ipk?

Mac0ne16, 2021-08-24:
> > > 看着需求比较少 就从胖版里拿出来了 slim版还是有预置的
> > 
> > 
> > 难过的是，我自己安装了好像还不能用。
> 
> 你怎么装的？slim从软件源安装还是第三方下载的ipk?

我也有同样的问题在这里下载https://github.com/rufengsuixing/luci-app-zerotier/releases/download/1.0-16/luci-app-zerotier_1.0-16_all.ipk 直接终端下安装，日志里看到缺少一个库也安装了，但是最还是报一个什么文件找不到，但是去源里搜也找不到包含的包。

iamfuhui, 2021-08-25:
> > > 看着需求比较少 就从胖版里拿出来了 slim版还是有预置的
> > 
> > 
> > 难过的是，我自己安装了好像还不能用。
> 
> 你怎么装的？slim从软件源安装还是第三方下载的ipk?

我之前一直是胖版，在线升级。我发现没有zerotier，去软件包里面找了，安装是报错的。

calkchu, 2021-08-30:
zerotier,我用slim版本安装了,而且我试了很多遍,有一个共性的问题,就是第一次安装后是可以使用的,之后,不知道什么原因,就无法使用了,zerotier官网也能监测到我这个路由器上的zerotier客户端,能够识别到IP,mac和版本,但是之后就是无法通信,平不同zerotier分配的同一网段其它地址,之前是没有这个问题的.

iamfuhui, 2021-08-30:
> zerotier,我用slim版本安装了,而且我试了很多遍,有一个共性的问题,就是第一次安装后是可以使用的,之后,不知道什么原因,就无法使用了,zerotier官网也能监测到我这个路由器上的zerotier客户端,能够识别到IP,mac和版本,但是之后就是无法通信,平不同zerotier分配的同一网段其它地址,之前是没有这个问题的.

作者已经将zerotier加入了胖版固件。

calkchu, 2021-08-30:
> > zerotier,我用slim版本安装了,而且我试了很多遍,有一个共性的问题,就是第一次安装后是可以使用的,之后,不知道什么原因,就无法使用了,zerotier官网也能监测到我这个路由器上的zerotier客户端,能够识别到IP,mac和版本,但是之后就是无法通信,平不同zerotier分配的同一网段其它地址,之前是没有这个问题的.
> 
> 作者已经将zerotier加入了胖版固件。

我原来使用slim版本装的，你说的胖版固件是slim吗？我的版本是0819的版本，是有zerotier的插件的，今晚上试试0830的版本

iamfuhui, 2021-08-30:
> > > zerotier,我用slim版本安装了,而且我试了很多遍,有一个共性的问题,就是第一次安装后是可以使用的,之后,不知道什么原因,就无法使用了,zerotier官网也能监测到我这个路由器上的zerotier客户端,能够识别到IP,mac和版本,但是之后就是无法通信,平不同zerotier分配的同一网段其它地址,之前是没有这个问题的.
> > 
> > 
> > 作者已经将zerotier加入了胖版固件。
> 
> 我原来使用slim版本装的，你说的胖版固件是slim吗？我的版本是0819的版本，是有zerotier的插件的，今晚上试试0830的版本

我指的是 r2s.img.gz 这个版本，非 slim 版本。