stupidloud/nanopi-openwrt #820: 0827升级0828报错，can't unmount /dev/loop0p2: Invalid argument

blueboyggh, 2021-08-28:
The partition table has been altered.
Syncing disks.
解压已完成，准备编辑镜像文件，写入备份信息
Sat Aug 28 19:34:57 CST 2021 upgrade: Saving config files...
备份文件已经写入，移除挂载
umount: can't unmount /dev/loop0p2: Invalid argument

然后升级过程就终止了，我0827也是在线升级上来的，升级完了以后遇到了能正常用，但是system灯一直闪，而且后台进不去的问题，重启了一下好了，今天升级0828就升不上去了

我重新刷一下固件的话，以后在线升级能用吗？

blueboyggh, 2021-08-28:
是否是在线升级脚本不适配新的snapshot代码导致的？

xiaoming3039, 2021-08-29:
> 是否是在线升级脚本不适配新的场景代码导致的？

不是的 很久以前的版本升级 我就已经发现这个问题了

klever1988, 2021-08-29:
添加了卸载前判断，试试

xiaoming3039, 2021-08-29:
> 添加了卸载前判断，试试
具体怎么操作  不会阿

klever1988, 2021-08-29:
> > 添加了卸载前判断，试试
> > 具体怎么操作  不会阿

我是说我添加了。
你直接和以前一样执行升级命令就行

blueboyggh, 2021-08-29:
> 添加了卸载前判断，试试

额，升级流程走完了，但是怎么概况页面里的固件版本还显示0827呢？这到底是升级成了还是没成...

klever1988, 2021-08-29:
> > 添加了卸载前判断，试试
> 
> 额，升级流程走完了，但是怎么概况页面里的固件版本还显示0827呢？这到底是升级成了还是没成...

浏览器缓存问题，新的luci容易出这个毛病，ctrl+f5强制刷新

blueboyggh, 2021-08-29:
> > > 添加了卸载前判断，试试
> > 
> > 
> > 额，升级流程走完了，但是怎么概况页面里的固件版本还显示0827呢？这到底是升级成了还是没成...
> 
> 浏览器缓存问题，新的luci容易出这个毛病，ctrl+f5强制刷新

换了个浏览器好了，感谢大佬！

closed 2021-08-29 by blueboyggh