stupidloud/nanopi-openwrt #822: R4S 0828 Passwall不能用

eenima, 2021-08-28:
订阅完节点后 在基础设置中选择TCP节点后勾上主开关保存并应用
显示配置已应用之后无事发生 TCP显示未运行

Goodman-vip, 2021-08-28:
Passwall 分流早就不能用了，我刷回老版去了

Goodman-vip, 2021-08-28:
而且新版的各种问题

eenima, 2021-08-28:
我是根本用不了 OpenClash也是 刷了别的固件 端口转发又有问题

symantes, 2021-08-29:
我也是，也刷回了老版本. 另外，新版本界面不如老版本流畅，无论你怎么弄！

symantes, 2021-08-29:
我也刷回了25号老版本。另外，新版本不流畅，一跳一跳的，无论你怎么弄！

eenima, 2021-08-29:
看来还是得自己学习一下编译自己的固件

Goodman-vip, 2021-08-29:
我用回8.6号的版本了，我觉得目前最稳定

klever1988, 2021-08-29:
不能启动的原因太多了 得进终端用命令行启动啊

我这用r2s试了一下，passwall+xray+vless+tls没问题

eenima, 2021-08-29:
啊这 咱这些小白也不太会……

oldoldstone, 2021-08-30:
> 不能启动的原因太多了 得进终端用命令行启动才知道具体原因啊
> 
> 我这用r2s试了一下，passwall+xray+vless+tls没问题
r2s slim里没有xray-core
装ssr提示
Required dependency package xray-core is not available in any repository.

klever1988, 2021-08-30:
> > 不能启动的原因太多了 得进终端用命令行启动才知道具体原因啊
> > 我这用r2s试了一下，passwall+xray+vless+tls没问题
> > r2s slim里没有xray-core
> > 装ssr提示
> > Required dependency package xray-core is not available in any repository.

slim?

oldoldstone, 2021-08-31:
> > > 不能启动的原因太多了 得进终端用命令行启动才知道具体原因啊
> > > 我这用r2s试了一下，passwall+xray+vless+tls没问题
> > > r2s slim里没有xray-core
> > > 装ssr提示
> > > Required dependency package xray-core is not available in any repository.
> 
> slim?
搞错了，是docker版

klever1988, 2021-08-31:
[image]

预装了的

klever1988, 2021-08-31:
docker和胖版都自带ssr的，你还要自己装ssr... 不知道你在说什么