stupidloud/nanopi-openwrt #832: 设备上网问题

jiam9700, 2021-09-03:
用的是最新版的r2s固件, ipad和iphone上网是可以跑满300mbps带宽的, 电脑却只有100mbps (重启之后又恢复为300mbps, 过一段时间会被限速)... 就很奇怪

klever1988, 2021-09-03:
[image]

看看这边的网卡模式是百兆还是千兆的

jiam9700, 2021-09-04:
都是千兆全双工, 但是电脑时不时会降速, 手机和ipad不会

jiam9700, 2021-09-04:
[image]
[image]
这是同一时间段手机和电脑测速的截图, 我电脑上分别用edge 和chrome测了speedtest & speedtest.cn, 速度都不超过100mbps

jiam9700, 2021-09-04:
[image]

jiam9700, 2021-09-04:
[image]

jiam9700, 2021-09-04:
[image]

lhf19900402, 2021-09-04:
> [image]
> 
> 看看这边的网卡模式是百兆还是千兆的

[image]

新固件就设置了一个ipv6拨号上网，怎么首页出现了这么多上网协议😆

klever1988, 2021-09-04:
@jiam9700 你登陆连接r2s的那台无线路由看看吧  应该是这里被降成百兆了

jiam9700, 2021-09-04:
多谢大佬, 确实是我那台tuf ax3000翻车了, 重启电脑之后rx就有700mbps+了, 但是过段时间还是会被限速...