stupidloud/nanopi-openwrt #833: r2s 0903 更新后

iamfuhui, 2021-09-04:
通过命令更新了0903 r2s胖版后
status/overview
概览 
<html>
<body>

CPU 信息 | ?
-- | --

</body>
</html>

adgurad home 错误

> download success start copy
start copy
awk: /etc/AdGuardHome.yaml: No such file or directory
uci: I/O error
sed: can't create temp file '/etc/hostsXXXXXX': Read-only file system
/usr/share/AdGuardHome/addhost.sh: line 31: can't create /etc/hosts: Read-only file system
/usr/share/AdGuardHome/addhost.sh: line 32: can't create /etc/hosts: Read-only file system
AdGuardHome service disabled
Command failed: Not found
rm: can't remove '/usr/bin/AdGuardHome': Read-only file system
mv: can't remove '/usr/bin/AdGuardHome': Read-only file system
mv failed maybe not enough space please use upx or change bin to /tmp/AdGuardHome

klever1988, 2021-09-04:
内存卡的文件系统坏掉了  拔卡冲刷吧~

iamfuhui, 2021-09-04:
已重刷。

closed 2021-09-04 by iamfuhui