stupidloud/nanopi-openwrt #838: 无法挂载主控为asm235cm的移动硬盘盒

ilove125006436, 2021-09-06:
经测试无法挂载主控为asm235cm的移动硬盘盒，请问如何解决这个问题呢？

klever1988, 2021-09-06:
dmesg输出内容复制过来看看

ilove125006436, 2021-09-06:
> dmesg输出内容复制过来看看

[    0.095651] usbcore: registered new interface driver usbfs
[    0.096242] usbcore: registered new interface driver hub
[    0.096790] usbcore: registered new device driver usb
[    0.219925] ehci-platform ff5c0000.usb: EHCI Host Controller
[    0.220497] ehci-platform ff5c0000.usb: new USB bus registered, assigned bus number 1
[    0.221359] ehci-platform ff5c0000.usb: irq 29, io mem 0xff5c0000
[    0.236283] ehci-platform ff5c0000.usb: USB 2.0 started, EHCI 1.00
[    0.240394] ohci-platform ff5d0000.usb: Generic Platform OHCI controller
[    0.241017] ohci-platform ff5d0000.usb: new USB bus registered, assigned bus number 2
[    0.241899] ohci-platform ff5d0000.usb: irq 30, io mem 0xff5d0000
[    0.314036] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
[    0.316860] usbcore: registered new interface driver usb-storage
[    0.377226] usbcore: registered new interface driver usbhid
[    0.377717] usbhid: USB HID core driver
[    0.648415] usb 4-1: new SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[    1.976941] usbcore: registered new interface driver uas
[   10.710099] usbcore: registered new interface driver r8152
[   10.728390] usbcore: registered new interface driver ums-alauda
[   10.730374] usbcore: registered new interface driver ums-cypress
[   10.732140] usbcore: registered new interface driver ums-datafab
[   10.733934] usbcore: registered new interface driver ums-freecom
[   10.735710] usbcore: registered new interface driver ums-isd200
[   10.737497] usbcore: registered new interface driver ums-jumpshot
[   10.739157] usbcore: registered new interface driver ums-karma
[   10.741066] usbcore: registered new interface driver ums-sddr09
[   10.743020] usbcore: registered new interface driver ums-sddr55
[   10.744858] usbcore: registered new interface driver ums-usbat
[   10.747523] usbcore: registered new interface driver usblp
[   10.874391] usb 4-1: reset SuperSpeed Gen 1 USB device number 2 using xhci-hcd
[   10.964619] usbcore: registered new interface driver mt76x2u
[   11.292203] usbcore: registered new interface driver rtl88XXau
[   11.716692] usbcore: registered new interface driver rtl8821cu
[   12.251571] usbcore: registered new interface driver rtl88x2bu
[   12.298547] usbcore: registered new interface driver mt76x0u
[   12.311381] usbcore: registered new interface driver ath9k_htc

ilove125006436, 2021-09-06:
> dmesg输出内容复制过来看看

root@ImmortalWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.143 (runner@fv-az116-991) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r20038-2385ffcedb)) #0 SMP PREEMPT Thu Sep 2 20:13:17 2021
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s48152 r8192 d29672 u86016
[    0.000000] pcpu-alloc: s48152 r8192 d29672 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1003796K/1046528K available (8190K kernel code, 572K rwdata, 2628K rodata, 1728K init, 739K bss, 34540K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001599] Console: colour dummy device 80x25
[    0.002064] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002998] pid_max: default: 32768 minimum: 301
[    0.003687] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004365] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007352] ASID allocator initialised with 32768 entries
[    0.007994] rcu: Hierarchical SRCU implementation.
[    0.009722] smp: Bringing up secondary CPUs ...
[    0.010891] Detected VIPT I-cache on CPU1
[    0.010976] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.011765] Detected VIPT I-cache on CPU2
[    0.011826] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012603] Detected VIPT I-cache on CPU3
[    0.012664] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.012783] smp: Brought up 1 node, 4 CPUs
[    0.016081] SMP: Total of 4 processors activated.
[    0.016517] CPU features: detected: 32-bit EL0 Support
[    0.016986] CPU features: detected: CRC32 instructions
[    0.025479] CPU: All CPU(s) started at EL2
[    0.025948] alternatives: patching kernel code
[    0.042658] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.043560] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.044788] pinctrl core: initialized pinctrl subsystem
[    0.046486] NET: Registered protocol family 16
[    0.049290] DMA: preallocated 2048 KiB pool for atomic allocations
[    0.051154] cpuidle: using governor menu
[    0.051795] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers.
[    0.052658] Serial: AMBA PL011 UART driver
[... 270 lines cut ...]
[   11.427736] usbcore: registered new interface driver rtl88x2bu
[   11.428250] RTW: module init ret=0
[   11.469013] l2tp_ppp: PPPoL2TP kernel driver, V2.0
[   11.474213] usbcore: registered new interface driver mt76x0u
[   11.486829] usbcore: registered new interface driver ath9k_htc
[   11.487869] kmodloader: done loading kernel modules from /etc/modules.d/*
[   14.225780] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[   14.225780] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   18.095221] br-lan: port 1(eth1) entered blocking state
[   18.095705] br-lan: port 1(eth1) entered disabled state
[   18.096711] device eth1 entered promiscuous mode
[   18.099426] br-lan: port 1(eth1) entered blocking state
[   18.099909] br-lan: port 1(eth1) entered forwarding state
[... cut ...]
[   19.101260] br-lan: port 1(eth1) entered disabled state
[   19.947824] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   19.949276] r8152 4-1:1.0 eth1: carrier on
[   19.950358] br-lan: port 1(eth1) entered blocking state
[   19.950844] br-lan: port 1(eth1) entered forwarding state
[   19.951813] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   25.341426] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[   25.342247] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

klever1988, 2021-09-06:
没看到有用信息

你拔下来再插上去
然后再打印dmesg

ilove125006436, 2021-09-06:
> 没看到有用信息
> 
> 你拔下来再插上去
> 然后再打印dmesg

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.143 (runner@fv-az116-991) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r20038-2385ffcedb)) #0 SMP PREEMPT Thu Sep 2 20:13:17 2021
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s48152 r8192 d29672 u86016
[    0.000000] pcpu-alloc: s48152 r8192 d29672 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1003796K/1046528K available (8190K kernel code, 572K rwdata, 2628K rodata, 1728K init, 739K bss, 34540K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000008] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001599] Console: colour dummy device 80x25
[    0.002064] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.002998] pid_max: default: 32768 minimum: 301
[    0.003687] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004365] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007352] ASID allocator initialised with 32768 entries
[    0.007994] rcu: Hierarchical SRCU implementation.
[    0.009722] smp: Bringing up secondary CPUs ...
[    0.010891] Detected VIPT I-cache on CPU1
[    0.010976] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.011765] Detected VIPT I-cache on CPU2
[    0.011826] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012603] Detected VIPT I-cache on CPU3
[    0.012664] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.012783] smp: Brought up 1 node, 4 CPUs
[    0.016081] SMP: Total of 4 processors activated.
[    0.016517] CPU features: detected: 32-bit EL0 Support
[    0.016986] CPU features: detected: CRC32 instructions
[    0.025479] CPU: All CPU(s) started at EL2
[    0.025948] alternatives: patching kernel code
[    0.042658] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.043560] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.044788] pinctrl core: initialized pinctrl subsystem
[    0.046486] NET: Registered protocol family 16
[    0.049290] DMA: preallocated 2048 KiB pool for atomic allocations
[    0.051154] cpuidle: using governor menu
[... 284 lines cut ...]
[   18.099909] br-lan: port 1(eth1) entered forwarding state
[   18.166110] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[   18.176493] rk_gmac-dwmac ff540000.ethernet eth0: No Safety Features support found
[   18.177196] rk_gmac-dwmac ff540000.ethernet eth0: PTP not supported by HW
[   18.177945] rk_gmac-dwmac ff540000.ethernet eth0: configuring for phy/rgmii link mode
[   19.101260] br-lan: port 1(eth1) entered disabled state
[   19.947824] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   19.949276] r8152 4-1:1.0 eth1: carrier on
[   19.950358] br-lan: port 1(eth1) entered blocking state
[   19.950844] br-lan: port 1(eth1) entered forwarding state
[   19.951813] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   25.341426] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[   25.342247] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[... cut ...]
[ 1186.216916] sd 0:0:0:0: [sda] 30310400 512-byte logical blocks: (15.5 GB/14.5 GiB)
[ 1186.218318] sd 0:0:0:0: [sda] Write Protect is off
[ 1186.218750] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00
[ 1186.219513] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 1186.242325]  sda: sda1
[ 1186.245802] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 1186.518901] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[ 8608.143595] usb 1-1: USB disconnect, device number 2

klever1988, 2021-09-06:
[ 1186.216916] sd 0:0:0:0: [sda] 30310400 512-byte logical blocks: (15.5 GB/14.5 GiB)
[ 1186.218318] sd 0:0:0:0: [sda] Write Protect is off
[ 1186.218750] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00
[ 1186.219513] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
[ 1186.242325] sda: sda1
[ 1186.245802] sd 0:0:0:0: [sda] Attached SCSI removable disk
[ 1186.518901] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.

这不是识别了吗

ilove125006436, 2021-09-06:
> [ 1186.216916] sd 0:0:0:0: [sda] 30310400 512-byte logical blocks: (15.5 GB/14.5 GiB)
> [ 1186.218318] sd 0:0:0:0: [sda] Write Protect is off
> [ 1186.218750] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00
> [ [...]

那个是之前的16gU盘信息

ilove125006436, 2021-09-06:
> [ 1186.216916] sd 0:0:0:0: [sda] 30310400 512-byte logical blocks: (15.5 GB/14.5 GiB)
> [ 1186.218318] sd 0:0:0:0: [sda] Write Protect is off
> [ 1186.218750] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00
> [ [...]

root@ImmortalWrt:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.4.143 (runner@fv-az116-991) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r20038-2385ffcedb)) #0 SMP PREEMPT Thu Sep 2 20:13:17 2021
[    0.000000] Machine model: FriendlyElec NanoPi R2S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] cma: Reserved 8 MiB at 0x000000003f800000
[    0.000000] On node 0 totalpages: 261632
[    0.000000]   DMA32 zone: 4088 pages used for memmap
[    0.000000]   DMA32 zone: 0 pages reserved
[    0.000000]   DMA32 zone: 261632 pages, LIFO batch:63
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] percpu: Embedded 21 pages/cpu s48152 r8192 d29672 u86016
[    0.000000] pcpu-alloc: s48152 r8192 d29672 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 257544
[    0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 1003796K/1046528K available (8190K kernel code, 572K rwdata, 2628K rodata, 1728K init, 739K bss, 34540K reserved, 8192K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu:     RCU event tracing is enabled.
[    0.000000] rcu:     RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000]  Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.001636] Console: colour dummy device 80x25
[    0.002104] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000)
[    0.003037] pid_max: default: 32768 minimum: 301
[    0.003721] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.004400] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.007416] ASID allocator initialised with 32768 entries
[    0.008054] rcu: Hierarchical SRCU implementation.
[    0.009784] smp: Bringing up secondary CPUs ...
[    0.010947] Detected VIPT I-cache on CPU1
[    0.011033] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.011805] Detected VIPT I-cache on CPU2
[    0.011869] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.012613] Detected VIPT I-cache on CPU3
[    0.012672] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.012799] smp: Brought up 1 node, 4 CPUs
[    0.016115] SMP: Total of 4 processors activated.
[    0.016550] CPU features: detected: 32-bit EL0 Support
[    0.017019] CPU features: detected: CRC32 instructions
[    0.025507] CPU: All CPU(s) started at EL2
[    0.025973] alternatives: patching kernel code
[    0.042623] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.043530] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.044764] pinctrl core: initialized pinctrl subsystem
[    0.046466] NET: Registered protocol family 16
[    0.049305] DMA: preallocated 2048 KiB pool for atomic allocations
[... 275 lines cut ...]
[   13.110380] l2tp_ppp: PPPoL2TP kernel driver, V2.0
[   13.115625] usbcore: registered new interface driver mt76x0u
[   13.128320] usbcore: registered new interface driver ath9k_htc
[   13.129378] kmodloader: done loading kernel modules from /etc/modules.d/*
[   15.866221] xt_FULLCONENAT: RFC3489 Full Cone NAT module
[   15.866221] xt_FULLCONENAT: Copyright (C) 2018 Chion Tang <tech@chionlab.moe>
[   18.562509] br-lan: port 1(eth1) entered blocking state
[   18.563181] br-lan: port 1(eth1) entered disabled state
[   18.564057] device eth1 entered promiscuous mode
[   18.567074] br-lan: port 1(eth1) entered blocking state
[   18.567564] br-lan: port 1(eth1) entered forwarding state
[   18.634097] rk_gmac-dwmac ff540000.ethernet eth0: PHY [stmmac-0:01] driver [RTL8211E Gigabit Ethernet]
[... cut ...]
[   19.580775] br-lan: port 1(eth1) entered disabled state
[   20.223762] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   20.224714] r8152 4-1:1.0 eth1: carrier on
[   20.225910] br-lan: port 1(eth1) entered blocking state
[   20.226410] br-lan: port 1(eth1) entered forwarding state
[   20.227373] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   25.821478] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[   25.822361] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

这个是插上移动硬盘的r2s，拔下再插上的信息

klever1988, 2021-09-06:
> > [ 1186.216916] sd 0:0:0:0: [sda] 30310400 512-byte logical blocks: (15.5 GB/14.5 GiB)
> > [ 1186.218318] sd 0:0:0:0: [sda] Write Protect is off
> > [ 1186.218750] sd 0:0:0:0: [sda] Mode Sense: 45 00 00 00
> > [ 1186.219513] sd 0:0:0:0: [sda] Write cache: disabled, read cache: enabled, doesn't support DPO or FUA
> > [ 1186.242325] sda: sda1
> > [ 1186.245802] sd 0:0:0:0: [sda] Attached SCSI removable disk
> > [ 1186.518901] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
> > 这不是识别了吗
> 
> root@ImmortalWrt:~# dmesg
> [ 0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
> [ 0.000000] Linux version 5.4.143 (runner@fv-az116-991) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r20038-2385ffcedb)) #0 SMP PREEMPT Thu Sep 2 20:13:17 2021
> [ 0.000000] Machine model: FriendlyElec NanoPi R2S
> [ 0.000000] earlycon: uart8250 at MMIO32 0x00000000ff130000 (options '')
> [ 0.000000] printk: bootconsole [uart8250] enabled
> [ 0.000000] cma: Reserved 8 MiB at 0x000000003f800000
> [ 0.000000] On node 0 totalpages: 261632
> [ 0.000000] DMA32 zone: 4088 pages used for memmap
> [ 0.000000] DMA32 zone: 0 pages reserved
> [ 0.000000] DMA32 zone: 261632 pages, LIFO batch:63
> [ 0.000000] psci: probing for conduit method from DT.
> [ 0.000000] psci: PSCIv1.0 detected in firmware.
> [ 0.000000] psci: Using standard PSCI v0.2 function IDs
> [ 0.000000] psci: MIGRATE_INFO_TYPE not supported.
> [ 0.000000] psci: SMC Calling Convention v1.0
> [ 0.000000] percpu: Embedded 21 pages/cpu s48152 r8192 d29672 u86016
> [ 0.000000] pcpu-alloc: s48152 r8192 d29672 u86016 alloc=21_4096 [ 0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [ 0.000000] Detected VIPT I-cache on CPU0 [ 0.000000] CPU features: detected: ARM erratum 845719 [ 0.000000] Built 1 zonelists, mobility grouping on. Total pages: 257544 [ 0.000000] Kernel command line: console=ttyS2,1500000 earlycon=uart8250,mmio32,0xff130000 root=PARTUUID=5452574f-02 rw rootwait [ 0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear) [ 0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear) [ 0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off [ 0.000000] Memory: 1003796K/1046528K available (8190K kernel code, 572K rwdata, 2628K rodata, 1728K init, 739K bss, 34540K reserved, 8192K cma-reserved) [ 0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1 [ 0.000000] rcu: Preemptible hierarchical RCU implementation. [ 0.000000] rcu: RCU event tracing is enabled. [ 0.000000] rcu: RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4. [ 0.000000] Tasks RCU enabled. [ 0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies. [ 0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4 [ 0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0 [ 0.000000] GIC: Using split EOI/Deactivate mode [ 0.000000] random: get_random_bytes called from start_kernel+0x29c/0x3a0 with crng_init=0 [ 0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys). [ 0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns [ 0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns [ 0.001636] Console: colour dummy device 80x25 [ 0.002104] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=96000) [ 0.003037] pid_max: default: 32768 minimum: 301 [ 0.003721] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear) [ 0.004400] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear) [ 0.007416] ASID allocator initialised with 32768 entries [ 0.008054] rcu: Hierarchical SRCU implementation. [ 0.009784] smp: Bringing up secondary CPUs ... [ 0.010947] Detected VIPT I-cache on CPU1 [ 0.011033] CPU1: Booted secondary processor 0x0000000001 [0x410fd034] [ 0.011805] Detected VIPT I-cache on CPU2 [ 0.011869] CPU2: Booted secondary processor 0x0000000002 [0x410fd034] [ 0.012613] Detected VIPT I-cache on CPU3 [ 0.012672] CPU3: Booted secondary processor 0x0000000003 [0x410fd034] [ 0.012799] smp: Brought up 1 node, 4 CPUs [ 0.016115] SMP: Total of 4 processors activated. [ 0.016550] CPU features: detected: 32-bit EL0 Support [ 0.017019] CPU features: detected: CRC32 instructions [ 0.025507] CPU: All CPU(s) started at EL2 [ 0.025973] alternatives: patching kernel code [ 0.042623] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns [ 0.043530] futex hash table entries: 1024 (order: 4, 65536 bytes, linear) [ 0.044764] pinctrl core: initialized pinctrl subsystem [ 0.046466] NET: Registered protocol family 16 [ 0.049305] DMA: preallocated 2048 KiB pool for atomic allocations [ 0.051093] cpuidle: using governor menu [ 0.051728] hw-breakpoint: found 6 breakpoint and 4 watchpoint registers. [ 0.052596] Serial: AMBA PL011 UART driver [ 0.083222] HugeTLB registered 1.00 GiB page size, pre-allocated 0 pages [ 0.083845] HugeTLB registered 32.0 MiB page size, pre-allocated 0 pages [ 0.084450] HugeTLB registered 2.00 MiB page size, pre-allocated 0 pages [ 0.085060] HugeTLB registered 64.0 KiB page size, pre-allocated 0 pages [ 0.090288] sdmmc-regulator GPIO handle specifies active low - ignored [ 0.091004] reg-fixed-voltage sdmmc-regulator: Failed to register regulator: -517 [ 0.093414] iommu: Default domain type: Translated [ 0.095105] SCSI subsystem initialized [ 0.095762]  [... cut ...]
> [ 20.223762] r8152 4-1:1.0 eth1: Promiscuous mode enabled
> [ 20.224714] r8152 4-1:1.0 eth1: carrier on
> [ 20.225910] br-lan: port 1(eth1) entered blocking state
> [ 20.226410] br-lan: port 1(eth1) entered forwarding state
> [ 20.227373] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
> [ 25.821478] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
> [ 25.822361] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
> 
> 这个是插上移动硬盘的r2s，拔下再插上的信息

从启动到25秒都没有设备插入信息
你硬盘盒要上外部电源的吧

klever1988, 2021-09-06:
不用一直打全文  你就把设备插入 dmesg增加的那部分上下文发一下就行了 @ilove125006436

ilove125006436, 2021-09-06:
> 不用一直打全文 你就把设备插入 dmesg增加的那部分上下文发一下就行了 @ilove125006436

很奇怪，这款是绿联的asm235cm主控的2.5寸硬盘盒，线材是typec的，并没有外部供电，插电脑或者移动送的电视机顶盒都可以正常使用和识别。

ilove125006436, 2021-09-06:
> 不用一直打全文 你就把设备插入 dmesg增加的那部分上下文发一下就行了 @ilove125006436

移动电视机顶盒我刷了openwrt   可以正常挂载，自动挂载
系统

主机名 localhost 
主机型号 Generic sunxi board 
固件版本 OpenWrt Chaos Calmer 15.05 /LuCI 15.05-129-gc84aea1 Release (git-15.349.82008-7ad426e)  
内核版本 3.18.24_hi3798mv310 
本地时间 Mon Sep 6 05:15:56 2021 
运行时间 0h 4m 20s 
  

 
/dev/block/vold/8:1  

 
/android/mnt/sda/sda1  

 
150.82 GB / 150.91 GB  

 
0% (94.04 MB)  

 

/dev/block/vold/8:2  

 
/android/mnt/sda/sda2  

 
144.87 GB / 144.96 GB  

 
0% (93.83 MB)

ilove125006436, 2021-09-06:
> 移动电视机顶盒我刷了openwrt 可以正常挂载，自动挂载
> 系统
>
> 主机名 localhost
> 主机型号 Generic sunxi board
> 固件版本 OpenWrt Chaos Calmer 15.05 /LuCI 15.05-129-gc84aea1 Release (git-15.349.82008-7ad426e)
> 内核版本 3.18.24_hi3798mv310
> 本地时间 Mon [...]

机顶盒接口也是usb2.0的，可以正常识别和挂载，就是r2s不行，r2s接上硬盘有读碟声音，但是不是连续转动的声音

klever1988, 2021-09-06:
如果你接入硬盘盒但是dmesg没有打印出新增的日志，那就是r2s的usb口供电不足，无法启动你的硬盘

Freikys, 2021-09-06:
挂载硬盘建议使用4A的电源哦亲，这边建议您检查下您的电源呢亲。

ilove125006436, 2021-09-06:
> 如果你接入硬盘盒但是dmesg没有打印出新增的日志，那就是r2s的usb口供电不足，无法启动你的硬盘

是的，这个usb口供电可以更改设置吗？我刚换了oppo 5v 4a 的电源和线给r2s供电，仍然无法识别和挂载

ilove125006436, 2021-09-06:
> 挂载硬盘建议使用4A的电源哦亲，这边建议您检查下您的电源呢亲。

已经换了5v 4a 的oppo电源和线给r2s供电，仍然识别不到，硬盘有读盘声，但是只是滋滋的声音，硬盘不转动。插电脑和移动电视机顶盒刷的openwrt可以识别和挂载。

ilove125006436, 2021-09-06:
不知道这个和驱动有没有关系

klever1988, 2021-09-06:
r2s的usb电源输出上限肯定比你机顶盒啥的要低 所以你硬盘启动不起来
你用4a的电源只是提高了它的储备 并不能提高它的输出上限 
dmesg都没有设备连接信息 所以不是驱动的事情 还没有到这个层面

ilove125006436, 2021-09-06:
> r2s的usb电源输出上限肯定比你机顶盒啥的要低 所以你硬盘启动不起来
> 你用4a的电源只是提高了它的储备 并不能提高它的输出上限
> dmesg都没有设备连接信息 所以不是驱动的事情 还没有到这个层面

好的，谢谢，应给就是r2s u口供电不足的原因，另外一块成品的3.0usb移动硬盘可以识别和挂载

CNWindson, 2023-09-05:
> 移动电视机顶盒我刷了openwrt 可以正常挂载，自动挂载 系统
>
> 主机名 localhost 主机型号 Generic sunxi board 固件版本 OpenWrt Chaos Calmer 15.05 /LuCI 15.05-129-gc84aea1 Release (git-15.349.82008-7ad426e) 内核版本 3.18.24_hi3798mv310 本地时间 Mon [...]

老哥，海思这芯片咋刷openwrt呀，没找到教程，求助