stupidloud/nanopi-openwrt #843: 刷了近期几个版本的Docker默认都不能运行，还要怎么启动吗？

feeler2000, 2021-09-08:
都是恢复默认设置刷的，显示：Can NOT connect to docker daemon, please check!!

[image]

klever1988, 2021-09-08:
你这还是很早18k5.4分支的版本
我用今天的slim没有问题

[image]

feeler2000, 2021-09-08:
我以为只有r2s-with-docker.img.gz才带Docker，slim版本是不是要自己安装Docker的？

klever1988, 2021-09-08:
> 我以为只有r2s-with-docker.img.gz才带Docker，slim版本是不是要自己安装Docker的？

到软件包里，更新列表，把luci-i18n-dockerman-zh-cn点选安装就可以了

calkchu, 2021-09-15:
> > 我以为只有r2s-with-docker.img.gz才带Docker，slim版本是不是要自己安装Docker的？
> 
> 到软件包里，更新列表，把luci-i18n-dockerman-zh-cn点选安装就可以了
> 所有依赖都会自动解决

我在0915版本的 r2c slim固件里面按照dockerman，结果安装失败，显示如下图情况，不知道怎么能安装成功呀？

[image]

klever1988, 2021-09-15:
[image]
你卡的文件系统损坏了，拔到linux主机用e2fsck修复或者重刷

calkchu, 2021-09-15:
嗯嗯，已重刷解决了，0914版本无法用xshell传送文件，在线升级到0915也无法传送文件，08XX的版本没有这个问题

klever1988, 2021-09-15:
> 嗯嗯，已重刷解决了，0914版本无法用xshell传送文件，在线升级到0915也无法传送文件，08XX的版本没有这个问题

你用lrzsz传的吗 还是sftp？

calkchu, 2021-09-16:
> > 嗯嗯，已重刷解决了，0914版本无法用xshell传送文件，在线升级到0915也无法传送文件，08XX的版本没有这个问题
> 
> 你用lrzsz传的吗 还是sftp？

我用xshell自带的xftp传送的，如图，之前的版本就可以应用起来
[image]

calkchu, 2021-09-16:
我今天又试了几个版本0909版本之前的R2C固件都没有问题，可以用这个xftp传送文件，之后的版本一直到0915都显示这个错误

klever1988, 2021-09-16:
[image]

这边手动打开一下，
我默认给他关闭了
为了安全。

calkchu, 2021-09-17:
> [image]
> 
> 这边手动打开一下，
> 我默认给他关闭了
> 为了安全。

好像不是这个问题，我重刷了0916版本，安装了vsftpd，像你说的那样勾选了图中设置，但是用xftp依旧不行，跟没勾选是一样的，怀疑是不是需要重启，后来重启后服务什么的看了，都开启了，可是还是无法连接
[image]

calkchu, 2021-09-17:
我把Xftp里面的协议修改成ftp，才可以连接，原来是sftp，之前的版本是不需要这个调整的用sftp也可以进行连接，怎样可以应用sftp协议呢？