stupidloud/nanopi-openwrt #844: Slim版无法在线升级

wspeed0711, 2021-09-09:
有一段时间没更新了, 发现在动更新脚本停在了这几句:
esolving ghproxy.com... 146.56.175.58
Connecting to ghproxy.com|146.56.175.58|:443... connected.
Unable to establish SSL connection.
请帮我看看问题在哪里, 我是slim版.

klever1988, 2021-09-10:
opkg update
opkg install wget wget-ssl

wspeed0711, 2021-09-10:
root@ImmortalWrt:~# opkg update
Downloading file:///local_feed/Packages.gz
Updated list of available packages in /var/opkg-lists/local
root@ImmortalWrt:~# opkg install wget wget-ssl
Not downgrading package uclient-fetch on root from 2021-04-03-64e00d6d-1 to 1.21.1-1.
Package wget-ssl (1.21.1-1) installed in root is up to date.
似乎已经安装了这个包了, 最后仍然是之前的那个无法建立SSL连接.

klever1988, 2021-09-10:
你把xxx软件先停了再试试

wspeed0711, 2021-09-10:
多谢,果然是这个原因. 不过不得不吐槽下,新版界面太难看了

klever1988, 2021-09-10:
你可以切回argon啊 不过适配一般般就是了
但其实snapshot的luci用起来要比18.06的爽利 外观习惯一下就好