stupidloud/nanopi-openwrt #852: r2s的屏幕无法正常显示

mengxinz, 2021-09-14:
以前加装了屏幕还是希望用上这个功能，希望大佬修复一下

klever1988, 2021-09-14:
那个oled的屏幕？ 你装luci-app-oled插件了没。。

mengxinz, 2021-09-14:
> 那个oled的屏幕？ 你装luci-app-oled插件了没。。

你编译的固件里有oled,现在就是运行不起来

mengxinz, 2021-09-14:
> 那个oled的屏幕？ 你装luci-app-oled插件了没。。

刷入第一次开机显示正常，重启就不显示了状态设置不了
[image]

klever1988, 2021-09-14:
我这边一直都正常运行显示 你应该是某项设置搞错了 所以运行不起来

config oled
        option drawline '0'
        option drawrect '0'
        option fillrect '0'
        option drawcircle '0'
        option drawroundrect '0'
        option fillroundrect '0'
        option drawtriangle '0'
        option filltriangle '0'
        option displaybitmap '0'
        option displayinvertnormal '0'
        option drawbitmapeg '0'
        option date '1'
        option netspeed '0'
        option lanip '1'
        option cpufreq '1'
        option cputemp '1'
        option time '60'
        option enable '1'
        option from '0'
        option to '1440'
        option autoswitch '0'
        option scroll '1'
        option text 'OPENWRT'
        option netsource 'eth0'

mengxinz, 2021-09-14:
> 我这边一直都正常运行显示 你应该是某项设置搞错了 所以运行不起来
>
> ```
> config oled
>         option drawline '0'
>         option drawrect '0'
>         option fillrect '0'
>         option drawcircle '0'
>         option drawroundrect [...]

不会呀刷完我啥都没懂它就是这样，大小为119m的那个标准版固件

mengxinz, 2021-09-14:
> 我这边一直都正常运行显示 你应该是某项设置搞错了 所以运行不起来
>
> ```
> config oled
>         option drawline '0'
>         option drawrect '0'
>         option fillrect '0'
>         option drawcircle '0'
>         option drawroundrect [...]

之前摔过可能是屏坏了，不纠结了