stupidloud/nanopi-openwrt #855: frp客户端 不能正常链接

baojiliu, 2021-09-15:
* 固件版本	klever1988/nanopi-openwrt 2021-09-14

[image]
[image]

klever1988, 2021-09-15:
看下系统日志呐

baojiliu, 2021-09-16:
```
Wed Sep 15 10:24:34 2021 daemon.info frpc[8394]: 2021/09/15 02:24:34 [1;33m[W] [service.go:104] login to server failed: dial tcp 127.0.0.1:7000: connect: connection refused[0m
Wed Sep 15 10:24:34 2021 daemon.info frpc[8394]: dial tcp 127.0.0.1:7000: connect: connection refused
Wed Sep 15 10:24:34 2021 daemon.info procd: Instance frpc::instance1 s in a crash loop 6 crashes, 0 seconds since last crash
```
这个是昨天连接信息，我配置的是我的服务器。但是链接的确实```127.0.0.1:7000```

klever1988, 2021-09-16:
cat /etc/config/frpc 看一下frpc的真实配置内容是否和你提交的一致

baojiliu, 2021-09-17:
cat 是我在WEB上的配置。但是，在WEB点击保存后进行没有运行，重启后可以运行了。

baojiliu, 2021-09-24:
关于```frp```我测试到的问题

* 在创建```tcp```类型代理时，```本地IP```不能保存到配置文件```/etc/config/frpc```
* 客户端```frpc```版本```0.37.0```，不能和服务器```frps```版本```0.37.0```，正常工作。
```
2021/09/21 00:00:01 [D] [control.go:219] [b0ff49b88013c08e] new work connection registered
2021/09/21 00:00:06 [D] [service.go:422] Accept new mux stream error: invalid protocol version
```

klever1988, 2021-09-24:
要不我换成这个源，你再试试

https://github.com/kuoruan/luci-app-frpc

viogus, 2021-09-29:
2021-09-27的版本缺少frpc包，在安装luci-app-frpc后，缺少frpc程序
wget https://github.com/kuoruan/openwrt-frp/releases/download/v0.37.1-1/frpc_0.37.1-1_aarch64_generic.ipk
opkg install *.ipk
后解决

klever1988, 2021-09-29:
> 2021-09-27的版本缺少frpc包，在安装luci-app-frpc后，缺少frpc程序 wget https://github.com/kuoruan/openwrt-frp/releases/download/v0.37.1-1/frpc_0.37.1-1_aarch64_generic.ipk opkg install *.ipk 后解决

ok
我加上了

closed 2022-11-11 by baojiliu