stupidloud/nanopi-openwrt #859: r1s slim版使用

laikitleung, 2021-09-17:
剛試了下slim版，改良很多，內存使用也相對穩定了，wireguard，cifs mount，nfs mount , rsync等都能安裝使用。這就基本解決了我的問題。
目前碰到的問題是
1. ip6tables的raw支持沒有，也不能安裝
2. 缺少shadowsocks-libev-tunnel ， 可以從給出的源安裝，但是運行需要/usr/lib/libsodium.so.23沒有，只能將/usr/lib/libsodium.so.27改名拷貝一份才能運行。
繼續試用中。

klever1988, 2021-09-17:
我不知道怎么给ip6tables添加raw表支持
shadowsocks-libev-tunnel应该是没有再内置的源里，我看了一下并没有预编译，现在已经加入预编译了，下次再试试。另外shadowsocks-libev-tunnel依赖libsodium，应该是有libsodium这个库的

laikitleung, 2021-09-17:
kmod-ipt-raw6

klever1988, 2021-09-17:
[image]

laikitleung, 2021-09-17:
感恩感恩，自用dns用ss-tunnel轉發一下就行了，还是很便捷的

laikitleung, 2021-09-20:
一个小问题，通过腾讯源安装luci-app-statistics , 启动无法创建对应的collectd.conf，导致图表不会显示。小白不明就里，手动创建一个/usr/bin/stat-genconfig >/etc/collectd.conf恢复显示