stupidloud/nanopi-openwrt #861: 请教一个非专业问题

lhf19900402, 2021-09-18:
[image]
请问这个出处到底在哪？能说详细点吗  😆

heningcai, 2021-09-24:
> [image]
> 
> 请问这个出处到底在哪？能说详细点吗 😆

同问

lhf19900402, 2021-09-24:
> > [image]
> > 请问这个出处到底在哪？能说详细点吗 😆
> 
> 同问

我后来摸索到了  也在Actions里面

heningcai, 2021-09-24:
> > > [image]
> > > 请问这个出处到底在哪？能说详细点吗 😆
> > 
> > 
> > 同问
> 
> 我后来摸索到了 也在Actions里面

请问具体怎么点呀，我没看见呀

lhf19900402, 2021-09-25:
> > > > [image]
> > > > 请问这个出处到底在哪？能说详细点吗 😆
> > > 
> > > 
> > > 同问
> > 
> > 
> > 我后来摸索到了 也在Actions里面
> 
> 请问具体怎么点呀，我没看见呀

[image]
[image]
[image]
[image]

heningcai, 2021-09-25:
> > > > > [image]
> > > > > 请问这个出处到底在哪？能说详细点吗 😆
> > > > 
> > > > 
> > > > 同问
> > > 
> > > 
> > > 我后来摸索到了 也在Actions里面
> > 
> > 
> > 请问具体怎么点呀，我没看见呀
> 
> [image]
> [image]
> [image]
> [image]

谢谢你