stupidloud/nanopi-openwrt #863: 我安装的slim 2021-09-17版 softether自动断开，配置数据丢失

lyglj, 2021-09-18:
我安装的slim 2021-09-17版 softether自动断开，配置数据丢失，后台直接安装的插件5.0和那个非5.0的都安装了，配置好后可以使用，2 3分钟就自动就断开 好像是服务重启了，数据也被初始化了
系统日志中会有多条类似这种日志
Sat Sep 18 14:19:58 2021 daemon.notice softethervpn-server[9807]: Writing vpn_server.config (AutoSaveConfigSpan set to < 86400 > seconds)

lyglj, 2021-09-18:
我是R4S

calkchu, 2021-09-18:
> 我是R4S

你是怎么配置的，我正好在研究这个，不太会配置，有没有什么教程啥的可以发出来看看

lyglj, 2021-09-18:
就后台安装luci-i18n-softethervpn-zh-cn 然后用SE-VPN Server Manager (Tools) 连接管理

calkchu, 2021-09-18:
> 就后台安装luci-i18n-softethervpn-zh-cn 然后用SE-VPN Server Manager (Tools) 连接管理

是的，我就是这么配置的，但是配置完了以后iPhone连不上呀，用的l2tp协议

lyglj, 2021-09-18:
路由器的什么BBR 什么加速的都关掉 端口映射，我测试链接是用的softehter的客户端 还没到l2tp 这一步

calkchu, 2021-09-18:
哦哦，我这边没装TutboAAC，所以没有你说的那些加速，我也是参考网上的视频，说用l2tp可以在iPhone里面设置vpn连接，我试了不行，不知道还有没有其它方法可以连接

lyglj, 2021-09-18:
ios windows 肯定可以的 先内网测试

calkchu, 2021-09-18:
是条思路，回头我先内网测测看