stupidloud/nanopi-openwrt #865: 20210918很棒！

symantes, 2021-09-19:
20210918很棒！我从0826之前的版本升级上来了，速度很快！

symantes, 2021-09-19:
只是R4S的hello word、SSR+无法订阅

klever1988, 2021-09-19:
很可能是你的airport订阅格式特殊不被支持

symantes, 2021-09-19:
> 很可能是你的机场订阅格式特殊情况支持

相同的订阅，R2S没有问题。

klever1988, 2021-09-19:
> > 很可能是你的机场订阅格式特殊情况支持
> 
> 相同的订阅，R2S没有问题。

ssrp有啥日志输出么

symantes, 2021-09-19:
> > > 很可能是你的机场订阅格式特殊情况支持
> > 
> > 
> > 相同的订阅，R2S没有问题。
> 
> ssrp有啥日志输出么

2021-09-19 19:51:13 订阅失败, 恢复服务
2021-09-19 19:51:05 更新失败，没有可用的节点信息
2021-09-19 19:51:05  xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx2 获取内容为空
2021-09-19 19:51:13: -----------end------------
2021-09-19 19:51:09: 广告过滤加载完毕。
2021-09-19 19:51:05: 未启动主节点，广告过滤正在加载。
2021-09-19 19:51:05: 禁止连接的域名加载完毕。
2021-09-19 19:51:05: 未启动主节点，禁止连接的域名正在加载。
2021-09-19 19:51:05: ----------start------------
2021-09-19 19:51:05   xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx1: 获取内容为空
2021-09-19 19:51:01 服务正在暂停

klever1988, 2021-09-19:
在路由器ssh终端里直接
curl http://订阅地址
看看有没有内容输出

symantes, 2021-09-19:
有内容输出，很长一串字母

klever1988, 2021-09-19:
你分别勾选“通过路由器自身代理更新订阅”都试试看

iamfuhui, 2021-09-21:
订阅这个问题，我r2s需要关闭HelloWorld的代理才能更新订阅成功。