stupidloud/nanopi-openwrt #879: 求助：R4S 安装NFS出问题

KenZhengV, 2021-09-28:
root@R4S:/# opkg install kmod-fs-nfs
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Unknown package 'kmod-fs-nfs'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-nfs-common found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-dnsresolver found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-nfs found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-fs-nfs.
root@R4S:/# opkg install kmod-fs-nfs-v3 --nodeps
Multiple packages (libgcc1 and libgcc1) providing same name marked HOLD or PREFER. Using latest.
Unknown package 'kmod-fs-nfs-v3'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-nfs-common found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-dnsresolver found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-nfs found, but incompatible with the architectures configured
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-fs-nfs-v3 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-fs-nfs-v3.

klever1988, 2021-09-28:
用slim

KenZhengV, 2021-09-29:
> 用slim

在线升级到slim会掉之前的配置吗？谢谢！

klever1988, 2021-09-29:
不会 不过你平时用的软件需要手动装就是了