stupidloud/nanopi-openwrt #883: 状态概览报错，求大神帮忙看看原因

lfdfxl, 2021-10-03:
SyntaxError
Unexpected identifier
  in http://192.168.123.1/luci-static/resources/view/status/include/10_system.js?v=2.2.4:?
  at http://192.168.123.1/luci-static/resources/luci.js?v=git-21.273.05721-4244f3b:181:29

[image]
重新登录、清理缓存和重启设备都无法解决。

mengxinz, 2021-10-03:
我还以为是我一个人出问题，害得我把数据全清了

mengxinz, 2021-10-03:
问题好像是从2号的版本开始的

klever1988, 2021-10-03:
修复了 等编译出固件试试

mengxinz, 2021-10-04:
> 修复了 等编译出固件试试

测试没问题了

yukkuna, 2021-10-04:
> > 修复了 等编译出固件试试
> 
> 测试没问题了

啊 我用argon主题还是会报错诶 openwrt2020不会

mengxinz, 2021-10-04:
> > > 修复了 等编译出固件试试
> > 
> > 
> > 测试没问题了
> 
> 啊 我用argon主题还是会报错诶 openwrt2020不会

> > > 修复了 等编译出固件试试
> > 
> > 
> > 测试没问题了
> 
> 啊 我用argon主题还是会报错诶 openwrt2020不会

我是slim版完全没问题

lfdfxl, 2021-10-04:
> 修复了 等编译出固件试试

感谢大佬。

closed 2021-10-04 by lfdfxl