stupidloud/nanopi-openwrt #903: ssr-plus插件特殊的工作模式[新]

klever1988, 2021-10-17:
[image]
这是我给ssrp插件添加的一种介于gfwlist和绕过cn两种方式之间的一种新的模式

gfwlist基于一套已收集的域名集合，当dnsmasq从指定的域名解析得到目标ip的时候，给ip打上标记，从而使得流量走代理
而绕过cn顾名思义就是判断连接的目标地址如果在cn的IP集合里，则直连，否则走代理

前一种方式存在侧漏，即有些域名已经被q却没有添加到gfwlist，
后一种模式存在过曝，你肯定不想下载p2p的时候国外流量也走代理，所以你得设定常用80/443端口的访问规则。

mosdns模式的效果类似绕过cn，但是并不是把所有的非cn IP都规定为走代理，而是在你浏览器打入域名的时候，mosdns解析得到ip地址并且判断是否cn，再对IP设定是否走代理的操作（所以称它“动态”）。这就避免了gfwlist的侧漏，也改进了绕过cn方式的机械。

所以，欢迎食用。。

（_使用这个模式时，不要介意ssrp的“状态”显示“DNS 防污染服务 未运行”，其实是被mosdns顶替了防污染的工作_）

kenyloveg, 2021-10-22:
大佬，有什么办法可以让SSR运行在全局模式下，本机的某些应用比如AdguardHome不走全局翻墙路线，谢谢

kenyloveg, 2021-11-01:
之前在大雕那里提过...
https://github.com/coolsnowwolf/lede/issues/6405

klever1988, 2021-11-01:
这里的mosdns已经可以实现你在那个issue提的需求，ssrp原有的“绕过”模式也可以实现。
ssrp在全局时dns肯定是经过代理的，但是你引入ros分流，除了给adg的上游dns加入白名单别无他法；
况且这个办法也并非你所说的“不现实啊”，因为一个范围更大的大陆ip都可以维护作为分流使用，怎么几个国外公共dns ip却变得无法维护了呢？

kid-oh, 2022-04-15:
但是启用mosdns动态模式后 状态选项里的连通性和数据库都无法连通和更新 不知道是否需要管它

ZhelinCheng, 2022-09-10:
大佬，我ssr PLUS+运行成功，但是无法访问外网是啥情况，我自己排查了许久，发现DNS解析不了外网网站，但我配置了谷歌DNS，我实在找不到方案了，求指引一下。
[image]

Shaco-Ma, 2024-05-20:
> 大佬，我ssr PLUS+运行成功，但是无法访问外网是啥情况，我自己排查了许久，发现DNS解析不了外网网站，但我配置了谷歌DNS，我实在找不到方案了，求指引一下。
> 
> 我下载的Openwrt x86原版镜像，自己安装的ssr plus+
> 
> [image]

大佬，你这个解决了吗?我发现我也是这样,nslookup也不无法解析...

ZhelinCheng, 2024-05-20:
> 的ssr plus+

@Shaco-Ma 我已经没有OpenWrt了，现在换成了ShellCrash，简单快捷。