stupidloud/nanopi-openwrt #922: 换成Argon主题后，登录路由器没反应

beautifulangle, 2021-10-30:
换成Argon主题后，登录路由器后可能能登录一两次。但是几次以后就没有任何反应，转一段时间的圈后，网页一片空白。

klever1988, 2021-10-30:
argon需要网络读取在线壁纸  如果你还没设置好网络 就可能卡在空白页

beautifulangle, 2021-10-31:
> argon需要网络读取在线壁纸 如果你还没设置好网络 就可能卡在空白页

感谢大大解答，在上传壁纸使用内建后直接就可以进入登录界面了，万分感谢

lcl-101, 2021-11-26:
怎么上传壁纸和使用内建了？我也碰到这个主题登录不了

beautifulangle, 2021-11-26:
有个argon主题设置，你进入选择内建然后在最下面有个选择图片上传，上传之后就不会有问题了

 

 
&mdash;

lcl-101, 2021-11-26:
我试试，我找到了背景图片路径 /www/luci-static/argon/background/

beautifulangle, 2021-11-26:
> 我试试，我找到了背景图片路径 /www/luci-static/argon/background/

> 
我都是每次换了主题马上就去上传壁纸😅
- 

1.

lcl-101, 2021-11-26:
argon主题设置,这个我没有找到在哪里，你的是在哪里了？嘿嘿

lcl-101, 2021-11-26:
[image]
是不是需要装一个这个，才能设置

beautifulangle, 2021-11-26:
肯定要啊
我用的是预装好的，系统栏目下，重启上面就是argon主题设置

lcl-101, 2021-11-26:
装上了，哈哈哈，解决，棒，哈哈哈，这个主题真的够了，哈哈

beautifulangle, 2021-11-26:
的确很漂亮

lcl-101, 2021-11-26:
对，很漂亮，不清楚作者为什么要留一个这个问题，哈哈哈

chenbeitang, 2021-12-16:
R2S用的其它固件，跑了一年了今天突然403，运气好看到了这个issue，解决了我的难题呀，感恩~

zhangyuheng, 2022-02-02:
官方问题链接 https://github.com/jerrykuku/luci-theme-argon/issues/266

tianyutingxy, 2022-05-05:
网上检索了一圈，居然碰到这个答案了，解决了，argon这是个什么sb设定。